PlotJuggler/PlotJuggler #15: libpng warning: bKGD: invalid

amdn, 2017-01-06:
On Fedora 25 libpng issues two warnings when PlotJuggler starts

```
$ ./PlotJuggler/build/bin/PlotJuggler 
libpng warning: bKGD: invalid
libpng warning: bKGD: invalid
```

I followed the fix suggested for this similar error, and the warning is gone:

http://stackoverflow.com/questions/22745076/libpng-warning-iccp-known-incorrect-srgb-profile

```
$ sudo dnf install ImageMagick
$ cd PlotJuggler
$ mogrify $(find . -name '*.png')
```

facontidavide, 2017-01-09:
Done! thank you for reporting

closed 2017-01-09 by facontidavide

w-azure, 2024-05-10:
> On Fedora 25 libpng issues two warnings when PlotJuggler starts
>
> ```
> $ ./PlotJuggler/build/bin/PlotJuggler
> libpng warning: bKGD: invalid
> libpng warning: bKGD: invalid
> ```
>
> I followed the fix [...]

I'm getting this error too, likely coming from my OpenCV. So when you used that command, does it delete the image or just remove the "RPG profile" (I don't know what that is)? Any idea what feature about a png image that causes it?