PlotJuggler/PlotJuggler #18: Repeated start and stop of streaming does sometimes freeze the output.

uliklank, 2017-01-10:
When plotting multiple ros topics, and starting and stopping the streaming, the program seems to fall into a mode, streaming can not activated successfully.

No obvious reset returns to a valid state.

facontidavide, 2017-01-10:
Is there any way I can recreate this problem locally on my machine?
Any suggestion?

Would you be able to take a screencast of the behavior using something like this ?

http://www.omgubuntu.co.uk/2013/12/simple-screen-recorder-linux

facontidavide, 2017-01-20:
I made few changes to the way that streaming coordinates with the main thread.

Can you pull the changes from master and tell me if you still experience the problem?

facontidavide, 2017-02-07:
Since I made several changes in the last weeks, I don't know if you still has the same problem or not...
I will temporarily close this issue. Don't hesitate to reopen it if the problem persists.

closed 2017-02-07 by facontidavide