PlotJuggler/PlotJuggler #22: Click and drag

scott-eddy, 2017-03-14:
I love the package for rosbag inspection.  Thanks for the work

Is there a way to click and drag the plot?  Additionally is there any way I can contribute to documentation?  I'd be happy to do a PR for the readme, but if you'd prefer a wiki etc I'd do that too.

facontidavide, 2017-03-14:
First of all, thanks a lot for contributing.

Dragging the rosbag would be complicated because the ROS related plugins have no access to the other widgets. 
There might be a solution though, I will think about it.

In terms of documentation, I guess I would prefer to extend to README.md or add other markdown documents together with the source code (there is a /doc folder)

scott-eddy, 2017-03-14:
Thanks for the update.  I think that click and drag would be an awesome feature for data analysis, it really helps viewing the time history on long running rosbags

facontidavide, 2017-03-14:
I start thinking that I missed what you meant.

I believed you was talking about drag and drop of files, but probably you don't.
Can you tell me what you mean by "click and drag"?

What do you want to click and where do you want to drag&drop it?

scott-eddy, 2017-03-14:
Gotcha,

I would like to be able to be zoomed in on a plot, click the mouse in the plot field and drag the entire plot with respect to time.  E.g. move the time axis left and right by moving my mouse.  Does that make sense?

facontidavide, 2017-03-14:
middle mouse button

scott-eddy, 2017-03-14:
Thanks, I'll put a PR to document this.  

For people on a laptop who don't have easy access to a middle mouse button it would be nice to have this be configurable via a preference.

facontidavide, 2017-03-14:
It is complicated :(

LEFT mouse does ZoomArea
LEFT + CTRL plot swap
LEFT + SHIFT move the vertical tracker
RIGHT mouse open the context menu

I guess I can add RIGHT + CTRL to be equivalent to  MIDDLE mouse ...

scott-eddy, 2017-03-14:
For laptop users MIDDLE mouse access with LEFT or RIGHT clicks is a key functionality.  However, I think that LEFT + CTRL would be better for MIDDLE mouse because RIGHT mouse + DRAG is often difficult on laptops whereas LEFT + DRAG is easy.  RIGHT mouse + CTRL as a discrete action for plot swap would be easy on a laptop as well.  

It would be my suggestion to have LEFT + CTRL be MIDDLE mouse to enable then DRAGGING the "MIDDLE" mouse, and map RIGHT + CTRL to plot swap as it doesn't require a drag.  

Obviously this is my preference and a change would effect current users but think about it.

facontidavide, 2017-03-14:
It make sense. Unfortunately laptop users would also miss zoom with middle wheel, that is the easiest way to zoom out...

scott-eddy, 2017-03-14:
That makes sense, though typically new trackpads have a pretty easily accessible MIDDLE scroll feature which shouldn't be effected by this change.  For example I'm on a macbook pro and two finger scrolling makes zooming in/out easy, but I have no easy way to depress the MIDDLE mouse button. I've also had trackpads that have MIDDLE scroll on the far left/right of the pad.

facontidavide, 2017-03-14:
Ok. I changed the mouse + modifiers combination to exclude middle wheel and middle button as much as possible.

I will close the issue

closed 2017-03-14 by facontidavide

scott-eddy, 2017-03-15:
CTRL + Left mouse brings up the drag icon for me but does not alter the curve

facontidavide, 2017-03-15:
my bad, during the merge of another branch I overwritten the feature