PlotJuggler/PlotJuggler #23: Remove data from plot

scott-eddy, 2017-03-14:
Is it possible to remove data from a plot once it has been added?  

If not this would be a great feature

scott-eddy, 2017-03-14:
Additionally to be able to hide data would be great without removing it

facontidavide, 2017-03-14:
Right now you can remove all the loaded data at once.
But I will think about removing improving this part (i.e. removing only the data selected in the left list).

You can not hide data, but you can select the one you are interested to using the "search" editor on the top-left area. That should be sufficient to easily filter out what you don't need, I think.

scott-eddy, 2017-03-14:
Thanks, for me it was not very intuitive to have to right click on the map to select which curves to remove.  But it works great.

facontidavide, 2017-03-14:
I just pushed this https://github.com/facontidavide/PlotJuggler/commit/408423a031180d28c152572b0ccb6c64ffee1a07

If you believe that right click is not intuitive (I also agre that it is a little bit too "easy"), what do you suggest? May be a command in the upper menu?

scott-eddy, 2017-03-14:
Thanks, I will try 408423 sometime today or tomorrow and let you know what I think.  I think that a separate button in the upper menu, or lower menu, that brings you to the selection pane would be great.

Additionally I think that curves currently being plotted should be highlighted  in the left pane.  Then you could right click on a highlighted curve and select "remove from plot" on each one.  But that would likely be a pretty large change

scott-eddy, 2017-03-14:
@facontidavide 408423a causes a segfault on my machine.  Below is the output:
```
"libDataLoadCSV.so" : is a DataLoader plugin 
"libDataLoadROS.so" : is a DataLoader plugin 
"libDataStreamROS.so" : is a DataStreamer plugin 
"libDataStreamSample.so" : is a DataStreamer plugin 
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used. 
"libRosTopicPublisher.so" : is a StatePublisher plugin 
"PlotJuggler" :  "Plugin verification data mismatch in '/home/eddy/OpenSource/ros_packages/devel/lib/plotjuggler/PlotJuggler'" 
The loading operation took 1267 milliseconds 
Segmentation fault (core dumped)
```

Reopening so this can be tracked

facontidavide, 2017-03-14:
Fixed. Additionally, instead of the context menu (showed by right click), you can use the DEL button (much more intuitive).

closed 2017-03-14 by facontidavide