PlotJuggler/PlotJuggler #24: Data name tab completion

scott-eddy, 2017-03-14:
It would be great to have the ability to tab complete data for rostopics that have a lot of subfields.

facontidavide, 2017-03-14:
Usually the search box works pretty well for me. I can easily navigate through in 500+ variables.

I might consider to add a third way to use the search box, n which a prefix (with tab completion) is used.

I honestly find this slower, but it can be done; let me think about it.

facontidavide, 2017-03-24:
Done!

You have an alternative way to filter the list on the left using the mode "prefix" (visible clicking on the cog icon).

When prefix mode is activated, you will see the autocompletion. Don't use TAB, though, I was not able to make it work

closed 2017-03-24 by facontidavide