PlotJuggler/PlotJuggler #28: [Feature Request] Clicking point shows x,y value of data ala MATLAB

scott-eddy, 2017-03-15:
@facontidavide I hope you don't get overwhelmed by all these issues, I just want to put ideas on paper so they are there for the future.   Also feel free to tell me what you won't prioritize so I can make PRs where you don't have bandwidth.

When viewing data with points user should be able to click on a specific point and see its exact x,y value.

facontidavide, 2017-03-15:
No problem, I do appreciate that someone care enough about PlotJuggler to contribute to it with suggestions :D

The vertical tracker triggered by SHIFT + LEFT works pretty well for me and the LEFT mouse without modifiers is already used to zoom an area.

For the time being I will not implement this feature.

facontidavide, 2017-03-21:
This is not exactly what you suggested, but is very similar.
https://github.com/facontidavide/PlotJuggler/commit/f3d7fe75a6182e6f0ed5cab3fe008475bccf9471

Now you will see the X,Y coordinates when you LEFT click (no modifier).

I will close the issue, because I am not planning to implement the feature you described for the time being.

closed 2017-03-21 by facontidavide