PlotJuggler/PlotJuggler #31: Feature request: Clear line data

peci1, 2017-03-28:
Thanks for the awesome package. One thing that I'd consider useful would be selectively clearing data of a single line (though not removing the "line object" itself), so that I can replay a bagfile and have the line data only from the last replay.

facontidavide, 2017-04-01:
Let me rephrase it to see if I understand correctly what you mean.

You want to keep the line object (the term "curve" is used int ot he source code, therefore let's use that), i.e. the graphical representation.

This curve is always associated to a certain data stored in memory (and visualized in the list on the left side of the screen.

You want to load something that share the same name, but you don't want it to overwrite the old one.
This is currebtly impossible and can't be implemente in the way you described.

NEVERTHELESS, a solution I see is to provide a prefix to the name during the loaginf phase to keep the data separated. This is the low hanging fruit for me.

What do you think?

peci1, 2017-04-01:
Well, what you've described sounds more like #32.

What I meant here is the following:
I'm running a ROS-enabled simulator which publishes some data. I subscribe to these data for a while, view them in PlotJuggler, and then I change something, restart the simulator, and want to see the new data (and only the new data).

What happens now is that the data from the first run of the simulator are still displayed and get mixed with the data from the second run. What I want is to remove data from the first run, but keep the "curve specifications and settings" as they were. Ideally without having to delete the curves.

A UI I'd propose is another item in the context menu of a plot called "Clear curve data", which would empty the datasets connected to the displayed curves, but not the curves themselves. The easier way is to always delete all data from the plot. The more difficult (but also more fruitful) option is to show a dialog similar to "remove curves" to let the user select which curves should get cleared.

I hope I explained it better this time.

facontidavide, 2017-04-02:
I guess that what you  call "curve specifications and settings" is what I call "layout", i.e. the way curves are arranged into differente widgets and tabs.
Correct me if I am wrong.

> I'm running a ROS-enabled simulator which publishes some data. I subscribe to these data for a while, view them in PlotJuggler, and then I change something, restart the simulator, and want to see the new data (and only the new data).

So you want something to clear the buffered data, but not the containers... I see. It is indeed possible but there is another solution. I can implement that easily in the streaming options.

Have you tried to save the layout ? You can **Delete all data** and then load the saved layout.

[image]

facontidavide, 2017-04-02:
Done (?)
Is this what you meant? https://github.com/facontidavide/PlotJuggler/commit/624836f928884ecf5974ea6859265fe839fa0a1d

closed 2017-04-02 by facontidavide

peci1, 2017-04-03:
Perfect, 624836f does exactly what I wanted. Thanks!