PlotJuggler/PlotJuggler #32: Compare data from bag/realtime?

peci1, 2017-03-28:
Is it somehow possible to show data from a topic loaded from a bagfile, and at the same time show data from a topic with the same name recorded realtime over ROS? I've seen the "rename fields" feature, but no "rename topics" feature.

facontidavide, 2017-04-01:
This can be solved as described in #31

facontidavide, 2017-04-02:
Ok, I got it now.

even if this is technically possible to rename the whole data using a prefix, I see a potential problem with the timestamp of the timeseries, 

You want to compare data in the past (rosbag) with more recent data (streaming).
When you try to comprare them, you will see an annoying offset in time that will be pretty messy.

Honestly I can't find right now an elegant way to  solve this problem

peci1, 2017-04-03:
You're right that's gonna be difficult. But I think PlotJuggler somehow remembers the first timestamp seen in the data, doesn't it? Then it could remember this offset per data source, which could solve the issue, couldn't it?

facontidavide, 2017-04-07:
I will keep this in the back of my mind, but I am afraid this would add a lot of complexity and corner cases in the code. Since I am trying to get the project stable for the release 1.0, I prefer not to include this feature for the time being. Maybe in the future.

Feel free of course to work on it and get the changes merged with a PR ;)

facontidavide, 2018-08-16:
With the latest version of PlotJuggler you can add a prefix to the name of a loaded or streamed topic.

This prevent data from rosbag and straming to overwrite each other.

Unfortunately, the time offset can NOT match and it would be absolutely the wrong thing to do to create arbitrary and inconsistent time offsets to address your needs. Therefore I decided that this is and will remain a "won't fix".

Cheers

closed 2018-08-16 by facontidavide