PlotJuggler/PlotJuggler #34: "Compatible formats" save dialog option unclear

peci1, 2017-03-28:
At least on my Ubuntu, the "save plot to file" dialog suggests it can save the plot as "Compatible formats", but there is no listing of what formats are compatible. There is also no warning/error message when you give the file an unsupported extension.

[image]

facontidavide, 2017-03-28:
Hi,

which version are you using? head of master? 
On my computer It does visualize the alternatives (jpg/png/etc)

peci1, 2017-03-28:
I'm on git master.

facontidavide, 2017-04-01:
I can not reproduce it. I believe that if you just click on Compatible Format with the mouse, the full sting should become visible.

Futhermore, I am following the suggestions of the original documentation. http://doc.qt.io/qt-5/qfiledialog.html#setNameFilter

closed 2017-04-01 by facontidavide