PlotJuggler/PlotJuggler #37: Aborted (core dumped) on data load

reinzor, 2017-04-19:
I am using the newest kinetic debians:

        terminate called after throwing an instance of 'std::runtime_error'
      what():  Unexpected character after type and field 'Definition of custom_msgs/CustomMsg
    Stack trace (most recent call last):

    #31   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fe79b1c682f, in __libc_start_main
          Source "/build/glibc-t3gR2i/glibc-2.23/csu/../csu/libc-start.c", line 291, in __libc_start_main [0x7fe79b1c682f]
    #30   Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x468829, in main
    #29   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c09bbeb, in QCoreApplication::exec()
    #28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c093b49, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
    #27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c0ec7ce, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
    #26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fe79a85549b, in g_main_context_iteration
    #25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fe79a8553ef, in g_main_context_dispatch
    #24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fe79a855196, in g_main_context_dispatch
    #23   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c0ec3c2, in QEventDispatcherGlibPrivate::runTimersOnceWithNormalPriority()
    #22   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c098785, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
    #21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c09638a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
    #20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe79c98e515, in QApplication::notify(QObject*, QEvent*)
    #19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe79c98905b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    #18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe79c982121, in QAction::event(QEvent*)
    #17   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c0c5c00, in QObject::event(QEvent*)
    #16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe79c981e1e, in QAction::setIconVisibleInMenu(bool)
    #15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe79c981897, in QAction::activate(QAction::ActionEvent)
    #14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe79c97f411, in QAction::triggered(bool)
    #13   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe79c0c4d29, in QMetaObject::activate(QObject*, int, int, void**)
    #12   Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x4a39ee, in TabbedPlotWidget::sendTabToNewWindow(PlotMatrix*)
    #11   Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x47709b, in MainWindow::onActionLoadDataFile(bool)
    #10   Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x476601, in MainWindow::onActionLoadDataFileImpl(QString, bool)
    #9    Object "/opt/ros/kinetic/lib/plotjuggler/libDataLoadROS.so", at 0x7fe76cf77196, in DataLoadROS::readDataFromFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)
    #8    Object "/opt/ros/kinetic/lib/libros_type_introspection.so", at 0x7fe76cd3c698, in RosIntrospection::buildROSTypeMapFromDefinition(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    #7    Object "/opt/ros/kinetic/lib/libros_type_introspection.so", at 0x7fe76cd3c21a, in RosIntrospection::ROSMessage::ROSMessage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    #6    Object "/opt/ros/kinetic/lib/libros_type_introspection.so", at 0x7fe76cd3bd19, in RosIntrospection::ROSField::ROSField(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe79bb1b918, in __cxa_throw
    #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe79bb1b700, in std::terminate()
    #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe79bb1b6b5, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
    #2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe79bb1d84c, in __gnu_cxx::__verbose_terminate_handler()
    #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fe79b1dd029, in abort
          Source "/build/glibc-t3gR2i/glibc-2.23/stdlib/abort.c", line 89, in __GI_abort [0x7fe79b1dd029]
    #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fe79b1db428, in gsignal
          Source "/build/glibc-t3gR2i/glibc-2.23/signal/../sysdeps/unix/sysv/linux/raise.c", line 54, in __GI_raise [0x7fe79b1db428]
    Aborted (Signal sent by tkill() 26463 1000)
    Aborted (core dumped)

reinzor, 2017-04-19:
Also compiled from source using newest version of https://github.com/facontidavide/ros_type_introspection , still gives me a runtime exception:

    terminate called after throwing an instance of 'std::runtime_error'
      what():  Unexpected character after type and field 'Definition of custom_msgs/CustomMsg
    Stack trace (most recent call last):
    #31   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4ab395282f, in __libc_start_main
          Source "/build/glibc-t3gR2i/glibc-2.23/csu/../csu/libc-start.c", line 291, in __libc_start_main [0x7f4ab395282f]
    #30   Object "/home/rein/ros/kinetic/dev/devel/lib/plotjuggler/PlotJuggler", at 0x51263f, in main
          Source "/home/rein/ros/kinetic/dev/src/PlotJuggler/plotter_gui/main.cpp", line 119, in main [0x51263f]
            117:         splash.close();
            118:         w.show();
          > 119:         return app.exec();
            120:     }
            121:     else{
            122:         MainWindow w( parser );
    #29   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab4827beb, in QCoreApplication::exec()
    #28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab481fb49, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
    #27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab48787ce, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
    #26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f4ab2fe149b, in g_main_context_iteration
    #25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f4ab2fe13ef, in g_main_context_dispatch
    #24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f4ab2fe1196, in g_main_context_dispatch
    #23   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab48783c2, in QEventDispatcherGlibPrivate::runTimersOnceWithNormalPriority()
    #22   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab4824785, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
    #21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab482238a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
    #20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f4ab511a515, in QApplication::notify(QObject*, QEvent*)
    #19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f4ab511505b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    #18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f4ab510e121, in QAction::event(QEvent*)
    #17   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab4851c00, in QObject::event(QEvent*)
    #16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f4ab510de1e, in QAction::setIconVisibleInMenu(bool)
    #15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f4ab510d897, in QAction::activate(QAction::ActionEvent)
    #14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f4ab510b411, in QAction::triggered(bool)
    #13   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4ab4850d29, in QMetaObject::activate(QObject*, int, int, void**)
    #12   Object "/home/rein/ros/kinetic/dev/devel/lib/plotjuggler/PlotJuggler", at 0x5795c9, in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)
          Source "/home/rein/ros/kinetic/dev/build/PlotJuggler/plotter_gui/moc_mainwindow.cpp", line 279, in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) [0x5795c9]
            276:         case 13: _t->onActionLoadLayout(); break;
            277:         case 14: _t->onActionLoadLayoutFromFile((*reinterpret_cast< QString(*)>(_a[1])),(*reinterpret_cast< bool(*)>(_a[2]))); break;
            278:         case 15: _t->onActionLoadDataFile((*reinterpret_cast< bool(*)>(_a[1]))); break;
          > 279:         case 16: _t->onActionLoadDataFile(); break;
            280:         case 17: _t->onActionLoadDataFileImpl((*reinterpret_cast< QString(*)>(_a[1])),(*reinterpret_cast< bool(*)>(_a[2]))); break;
            281:         case 18: _t->onActionLoadDataFileImpl((*reinterpret_cast< QString(*)>(_a[1]))); break;
            282:         case 19: _t->onActionReloadDataFileFromSettings(); break;
    #11   Object "/home/rein/ros/kinetic/dev/devel/lib/plotjuggler/PlotJuggler", at 0x51aa42, in MainWindow::onActionLoadDataFile(bool)
          Source "/home/rein/ros/kinetic/dev/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 792, in MainWindow::onActionLoadDataFile(bool) [0x51aa42]
            790:     ui->actionLoadRecentDatafile->setText("Load data from: " + filename);
            791: 
          > 792:     onActionLoadDataFileImpl(filename, false );
            793: }
            794: 
            795: void MainWindow::importPlotDataMap(const PlotDataMap& new_data)
    #10   Object "/home/rein/ros/kinetic/dev/devel/lib/plotjuggler/PlotJuggler", at 0x51bb49, in MainWindow::onActionLoadDataFileImpl(QString, bool)
          Source "/home/rein/ros/kinetic/dev/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 933, in MainWindow::onActionLoadDataFileImpl(QString, bool) [0x51bb49]
            931:         PlotDataMap mapped_data = loader->readDataFromFile(
            932:                     filename.toStdString(),
          > 933:                     timeindex_name   );
            934: 
            935:         _last_load_configuration = timeindex_name;
[... cut ...]
    #2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f4ab42a984c, in __gnu_cxx::__verbose_terminate_handler()
    #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4ab3969029, in abort
          Source "/build/glibc-t3gR2i/glibc-2.23/stdlib/abort.c", line 89, in __GI_abort [0x7f4ab3969029]
    #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4ab3967428, in gsignal
          Source "/build/glibc-t3gR2i/glibc-2.23/signal/../sysdeps/unix/sysv/linux/raise.c", line 54, in __GI_raise [0x7f4ab3967428]
    Aborted (Signal sent by tkill() 29763 1000)
    Aborted (core dumped)

facontidavide, 2017-04-19:
hi, thanks for reporting.

I will need the message definition file (the .msg)

facontidavide, 2017-04-19:
alternativeli, the rosbag that cause the crash

reinzor, 2017-04-20:
I will try to recreate a small bag with a custom msg file today.

-Rein

facontidavide, 2017-04-20:
if you can just copy and paste the message definition, that would be easier.

If you look into your message definition file ( I guess it is called __custom_msgs/CustomMsg.h__), look for the string below

    template<class ContainerAllocator>
    struct Definition< ::custom_msg::CustomMsg_<ContainerAllocator> >
    {
           static const char* value()

reinzor, 2017-04-20:
template<class ContainerAllocator>
    struct Printer< ::custom_msgs::CustomMsg_<ContainerAllocator> >
    {
      template<typename Stream> static void stream(Stream& s, const std::string& indent, const ::custom_msgs::CustomMsg_<ContainerAllocator>& v)
      {
        s << indent << "header: ";
        s << std::endl;
        Printer< ::std_msgs::Header_<ContainerAllocator> >::stream(s, indent + "  ", v.header);
        s << indent << "target_id: ";
        Printer<uint32_t>::stream(s, indent + "  ", v.target_id);
        s << indent << "station_id: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.station_id);
        s << indent << "cruise_speed: ";
        Printer<float>::stream(s, indent + "  ", v.cruise_speed);
        s << indent << "max_speed_reached: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.max_speed_reached);
        s << indent << "accel_min: ";
        Printer<float>::stream(s, indent + "  ", v.accel_min);
        s << indent << "accel_max: ";
        Printer<float>::stream(s, indent + "  ", v.accel_max);
        s << indent << "accel_desired: ";
        Printer<float>::stream(s, indent + "  ", v.accel_desired);
        s << indent << "steering_wheel_position: ";
        Printer<float>::stream(s, indent + "  ", v.steering_wheel_position);
        s << indent << "steering_wheel_position_desired: ";
        Printer<float>::stream(s, indent + "  ", v.steering_wheel_position_desired);
        s << indent << "vehicle_mode: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.vehicle_mode);
        s << indent << "emergency_brake: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.emergency_brake);
        s << indent << "longitudinal_controller_mode: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.longitudinal_controller_mode);
        s << indent << "lateral_controller_mode: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.lateral_controller_mode);
        s << indent << "active_mode: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.active_mode);
        s << indent << "vehicle_turn_indicator_signal: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.vehicle_turn_indicator_signal);
        s << indent << "lane_crossing_detected: ";
        Printer<uint8_t>::stream(s, indent + "  ", v.lane_crossing_detected);
      }
    };

    Header header
    
    uint32 target_id
    uint8  station_id
    
    float32 cruise_speed
    bool    max_speed_reached
    
    float32 accel_min
    float32 accel_max
    float32 accel_desired
    
    float32 steering_wheel_position
    float32 steering_wheel_position_desired
    
    uint8 vehicle_mode
    bool  emergency_brake
    
    uint8 longitudinal_controller_mode
    uint8 lateral_controller_mode
    
    uint8 active_mode
    
    uint8 vehicle_turn_indicator_signal
    bool lane_crossing_detected

facontidavide, 2017-04-20:
you sent __struct Printer__ instead of __struct Definition__  ;)

I really don't see anything suspicious in the message definition, I am afraid I will do need a rosbag to replicate the error.

reinzor, 2017-04-20:
I will prepare a zip with the bag and the package that contains the definition

reinzor, 2017-04-20:
Hmm, tried to reproduce it with a small bag file with only the specific message but this time it does not crash :(. I will log a small bag file on the real system next week and see if the problem occurs again.

facontidavide, 2017-04-20:
Is there any message that contains a __custom_msgs::CustomMsg__ in one of the message fields? May be the problem is the "parent", not this type.

reinzor, 2017-04-21:
I think I have found the possible cause, my hypothesis:

We use custom serialization for a non-ros platform and on the ros platform, we construct the ROS message with the topic_tools::ShapeShifter:

    // Instantiate empty message
    topic_tools::ShapeShifter msg;
    
    // Set message type and md5 hash 
    msg.morph(hash, msg_type_it->second, "Definition of " + msg_type_it->second, "false");

Here, we do not properly fill the message definition. Could this be the cause?

facontidavide, 2017-04-21:
Bingo!

Yes, basically what __ros-type-introspection__ does is to parse the definition and do some "reverse engineering" to understand at runtime which field exist and how the raw message memory can be deserialized.

This explain as well the weird message in the exception.

Can you provide somehow this definition ?

facontidavide, 2017-04-21:
@reinzor 

I will be very happy to help you fixing this. You can ask me questions by email at davide.faconti@gmail.com

Nevertheless, it is obvious that it is neither a PlotJuggler nor ros-type-introspection problem, so I will close this issue.

Cheers and thanks for reporting

closed 2017-04-21 by facontidavide

reinzor, 2017-04-24:
Hi @facontidavide , thanks for all your help! I am going to fix this using `rosserial_python`, it exposes a rosservice interface to query the correct message definition on runtime for cpp!

facontidavide, 2017-04-24:
Just curious...

How was the message generated? Through a __.msg__ file or programatically some how?

At the end of the day, rosserial_python find the message definition __somewhere__. There is no reason why you can't find this information too.

reinzor, 2017-04-24:
We were using the topic_tools:

Setup:

    ros::AdvertiseOptions a(topic_name, 10, md5, message_type, "Definition of " + message_type);

Publish:

    msg.morph(md5, message_type, "Definition of " + message_type, "false");

The message type is parsed using a ROS parameter and we were first obtaining the md5 via a system call `rosmsg md5 [message_type]` and just filling the definition with "Definition of " + message_type, which was ofcourse not correct but we didn't experience any problems until we analysed a bagfile with your tool ;)

facontidavide, 2017-04-24:
Yes, my approach is a little more pedantic and requires a "well formed" _topic_tools::ShapeShifter_ message :P

If you can share the entire file would actually help, if it is not super-secret (if you have privacy concern, send to my private email), otherwise I am afraid that I can hardly give you any support with the information you gave me.

For example, where does md5 come from? Usually it is coupled with the Definition info, so if you have one it is hard to believe that you don't have the other... (it is possible of course, but unusual).

reinzor, 2017-04-24:
Hi @facontidavide , I don;t think that is necessary, thanks. By querying the message_info service and filling the message properly, it will most probably work; if will let you know when we do some more testing! Thanks again!

reinzor, 2017-04-26:
Hi @facontidavide , just tested with the new set-up (proper message definitions), your tool works as a charm! Thanks!

facontidavide, 2017-04-26:
thank you for the update. Cheers