PlotJuggler/PlotJuggler #40: Axis settings

solosito, 2017-04-25:
I have been working with Plotjuggler and one thing I really miss is being able to set fix values for the axis. Would it possible to do it?

facontidavide, 2017-04-25:
I need you to be a little more specific.

By "fix values" in the axis do you mean the X axis (time), the Y value (value) or both?

Can't you just zoom?
Are you thinking about static data (load a rosbag) or streaming?

solosito, 2017-04-25:
I mean setting a min and a max value for Y axis. In both cases (static or streaming) would be nice.

The point is that now the Y axis is autosetting to the values of the data. Thus, for instance, if you get a big value for a sample at the beginning and the next one is very small and it keeps oscillating between around this value, you are not able to perceive the changes on it.

So if you are able to set the Y axis between two desired values you will able to see the data in the range you want even tough the bigger values are out of the "plotting window".

facontidavide, 2017-04-25:
Ok, I guess you are talking about __streaming__.

A feature like this is more complicated that it looks because I have to decide:

1) Is this setting part of the UNDO/REDO stack?
2) Can it be saved as a Layout.
3) How is it applied by the user? right click on the PlotWidget area?

I will find about it...

solosito, 2017-04-25:
Right, for the streaming. I forgot to say, sorry. 

In my opinion:

1. I really don't think it is necessary (i.e. as zooming).
2. This would be nice in order to just open the layout a plot directly without setting anything else.
3. Right click > Set Axis > Dialog for entering values as remove curves.
4. Having it in both axis would be helpful too for XY plot.

If you want I would be really glad to help you with it.

facontidavide, 2017-04-25:
> If you want I would be really glad to help you with it.

Thanks !!

I will create a new branch to develop this feature, I will let you know once it is available. As I said, it is more complicated that it looks like, IN PARTICULAR if you want to lock the X axis range, since a lot of logic depends on it.

solosito, 2017-04-25:
For the moment with the Y axis should be enough. I had to plot the data with another library everytime I had this problem.

Anyway, congratulations for your amazing tool :)

facontidavide, 2017-04-25:
Hi,

I have a branch that seems to work, even if the integration with UNDO and save layout need to be done.

https://github.com/facontidavide/PlotJuggler/tree/issue_40

Please check and let me know if you see anything weird or if you find a bug.
If you do, please send a screenshot or a video recording

Davide

solosito, 2017-04-25:
Somehow it is not working properly when the actual data takes a value out of the plotting range and the limits are changing. Also the numbers in the axis are being displayed incorrectly (increasing in the wrong order):

[image]

I am also uploading a video showing the behaviour.

solosito, 2017-04-25:
The behavior in the video it is exactly the expected from ~0:30 to ~1:30. The problems I described can be seen before and after that part.

https://youtu.be/T0VM4wxevPk

facontidavide, 2017-04-26:
I can fix the issue, but I am realizing that maybe what we actually want is not a minimum and maximum boundary for the Y axis, but instead FIXED values of the bottom and top of axis Y. 

That would be more intuitive, I think

solosito, 2017-04-26:
Exactly, being able to set fixed values for having the desired plotting range in order to be able to indirectly decide the granularity.

facontidavide, 2017-04-26:
bust pushed a new commit. take a look

facontidavide, 2017-04-26:
I think it is done (merged with master). Please tell me if there is any problem

closed 2017-04-26 by facontidavide

facontidavide, 2017-05-01:
Hi,

can you confirm that you are happy with the current version of the software? Have you detected any issue / corner case?

solosito, 2017-05-01:
Sorry, I couldn't answer before. It works as a charm. Thanks again!

ClemensVolk, 2022-02-23:
Hi,
I am using the ros-foxy version and can't figure out how to fix the scaling of the y-axis to [-1, 1]. It automatically re-scales to the data points.
Is this feature still active?

facontidavide, 2022-02-23:
right click->first option->right bottom corner of the dialog

ClemensVolk, 2022-02-24:
Thanks for the quick reply!
However it segfaults when either saving own values for "Max" & "Min" or the default ones.

ClemensVolk, 2022-02-24:
Opened up an Issue [Segmentation fault when saving axis settings #612](https://github.com/facontidavide/PlotJuggler/issues/612#issue-1149028835)