PlotJuggler/PlotJuggler #41: Changing X axis curve not working

pvnkmrksk, 2017-05-02:
I am not sure if this is the right place to post this. I tried as per the manual, using right click drag to set the x axis variable. I tried with different files and streams, it just doesn't work. It plots it on the y axis and the overlay of "set as x axis" never appears. Am I doing something wrong or is this a bug? Is there anything else I need to provide to fix this?

facontidavide, 2017-05-02:
Thank you for reporting.

Are you using the version on master (1.0.4) ? Which OS? Please give me as many details as possible

Please allow me to reproduce the bug, since this is not happening with master.

1) Launch the application with argument -t to generate some testing data.
2) Record your desktop with an application like this https://apps.ubuntu.com/cat/applications/quantal/gtk-recordmydesktop/
3) Send the video to me :)

facontidavide, 2017-05-04:
I have the feeling you may be using version 0.17, can you confirm?
The XY plot was added in version 1.0

facontidavide, 2017-05-06:
I didn't receive any update from you in 4 days. Please feel free to open again this ticket if you believe that the version on MASTER branch is affected by this issue.
For the time being, I will close the issue.

Cheers

Davide

closed 2017-05-06 by facontidavide