PlotJuggler/PlotJuggler #42: Crash loading bag with 2 topics of nav_msgs/Odometry

v-lopez, 2017-05-18:
- Version used 1.0.7 
- Attached two bags with the same topics, one is a few seconds long the other is longer.

The short one (named Error_opening.bag) crashes after selecting both topics inside the rosbag with the error:

> terminate called after throwing an instance of 'std::runtime_error'
  what():  can't deserialize this stuff: Header

The same happened when I tried the ROS Topic Streamer on those two topics at the same time.

The other bag is fine.

[bags.zip](https://github.com/facontidavide/PlotJuggler/files/1010678/bags.zip)

facontidavide, 2017-05-18:
Thanks for reporting, I will take a look ASAP

facontidavide, 2017-05-18:
There is something very weird related to the message with prefix __/ihmc__

Apparently the definition is incomplete. It looks like the message was created by hand and not from 
[nav_msgs/Odometry.h](http://docs.ros.org/diamondback/api/nav_msgs/html/Odometry_8h_source.html)

The other message instead is just fine.

Which process is publishing the the __valkyrie/output/robot_pose__?
This message can NOT be opened by Plotuggler.

facontidavide, 2017-05-18:
To be more specific. This is the **truncated** definition of the ihmc message

```
# This represents an estimate of a position and velocity in free space.  
# The pose in this message should be specified in the coordinate frame given by header.frame_id.
# The twist in this message should be specified in the coordinate frame given by the child_frame_id
Header header
string child_frame_id
geometry_msgs/PoseWithCovariance pose
geometry_msgs/TwistWithCovariance twist 
```

Whilst this it the **correct** one from /floating_base_pose

```
# This represents an estimate of a position and velocity in free space.  
# The pose in this message should be specified in the coordinate frame given by header.frame_id.
# The twist in this message should be specified in the coordinate frame given by the child_frame_id
Header header
string child_frame_id
geometry_msgs/PoseWithCovariance pose
geometry_msgs/TwistWithCovariance twist
================================================================================
MSG: std_msgs/Header
# Standard metadata for higher-level stamped data types.
# This is generally used to communicate timestamped data 
# in a particular coordinate frame.
# 
# sequence ID: consecutively increasing ID 
uint32 seq
#Two-integer timestamp that is expressed as:
# * stamp.sec: seconds (stamp_secs) since epoch (in Python the variable is called 'secs')
# * stamp.nsec: nanoseconds since stamp_secs (in Python the variable is called 'nsecs')
# time-handling sugar is provided by the client library
time stamp
#Frame this data is associated with
# 0: no frame
# 1: global frame
string frame_id

================================================================================
MSG: geometry_msgs/PoseWithCovariance
# This represents a pose in free space with uncertainty.

Pose pose

# Row-major representation of the 6x6 covariance matrix
# The orientation parameters use a fixed-axis representation.
# In order, the parameters are:
# (x, y, z, rotation about X axis, rotation about Y axis, rotation about Z axis)
float64[36] covariance

================================================================================
MSG: geometry_msgs/Pose
# A representation of pose in free space, composed of postion and orientation. 
Point position
Quaternion orientation

================================================================================
MSG: geometry_msgs/Point
# This contains the position of a point in free space
float64 x
float64 y
float64 z

================================================================================
MSG: geometry_msgs/Quaternion
# This represents an orientation in free space in quaternion form.

float64 x
float64 y
float64 z
float64 w

================================================================================
MSG: geometry_msgs/TwistWithCovariance
# This expresses velocity in free space with uncertainty.

Twist twist

# Row-major representation of the 6x6 covariance matrix
# The orientation parameters use a fixed-axis representation.
# In order, the parameters are:
# (x, y, z, rotation about X axis, rotation about Y axis, rotation about Z axis)
float64[36] covariance

================================================================================
MSG: geometry_msgs/Twist
# This expresses velocity in free space broken into its linear and angular parts.
Vector3  linear
Vector3  angular

================================================================================
MSG: geometry_msgs/Vector3
# This represents a vector in free space. 
# It is only meant to represent a direction. Therefore, it does not
# make sense to apply a translation to it (e.g., when applying a 
# generic rigid transformation to a Vector3, tf2 will only apply the
# rotation). If you want your data to be translatable too, use the
# geometry_msgs/Point message instead.

float64 x
float64 y
float64 z

```

v-lopez, 2017-05-18:
It is published by the IHMC Valkyrie Controller we are using in the Space Robotics Challenge.

Installation instructions are here: https://bitbucket.org/osrf/srcsim/wiki/system_setup

facontidavide, 2017-05-18:
As I said, the message seems to be corrupted and there is nothing I can do about it (a part from hard coding a solution for you).

I have tried to use rqt_plot and it is not able to extract the fields either.

facontidavide, 2017-05-18:
My suggestion is to ask the people a IHMC if they know why that message definition was truncated.
They are doing something "wrong" in my humble opinion.

v-lopez, 2017-05-18:
Yeah, certainly looks like that. Thanks for the help!

closed 2017-05-18 by v-lopez

facontidavide, 2017-05-18:
For the records.

**There is** a solution to solve this problem. I just tested it and it works;

You can **hard code** the message definition inside the plugin doing this:

```c++
#include <nav_msgs/Odometry.h>

DataLoadROS::DataLoadROS()
{
    _extensions.push_back( "bag");

    RosIntrospectionFactory::get().registerMessage(
                ros::message_traits::DataType<nav_msgs::Odometry>::value(),
                ros::message_traits::MD5Sum<nav_msgs::Odometry>::value(),
                ros::message_traits::DataType<nav_msgs::Odometry>::value(),
                ros::message_traits::Definition<nav_msgs::Odometry>::value() );
}
```

v-lopez, 2017-05-18:
Awesome, next time you come by PAL I'll invite you to a beer!