PlotJuggler/PlotJuggler #43: Feature request full screen mode

v-lopez, 2017-05-22:
Again awesome tool.

A nice to have feature would be to be able to remove most of GUI elements when my plot has been configured set.

Sometimes I would like to have a plot on a corner of my screen so I can work on other stuff while I am monitoring it.

My best approach was to create a new window for the plot, but even with that I'd say 40% or more of the screen space is used by widgets that at this moment are not needed.

My best approach is this:
[image]

I wish we could have something like this:
[image]

It's a minor feature, and if noone has done it by then, I will try to do it in july when I'll have more free time.

facontidavide, 2017-05-23:
I will keep it in mind. But I probably I will not start working on this feature anytime soon.

facontidavide, 2017-06-30:
F10 !!! let me know if it works OK for you

closed 2017-06-30 by facontidavide