PlotJuggler/PlotJuggler #46: CSV file handling

dmiklic, 2017-06-18:
First of all, thanks a lot for a great tool!

So far, I have been using PlotJuggler with rosbags and it has worked flawlessly. Now I'm trying to use it with `.csv` files and I can't quite figure out how the importing works. I'm using version 1.0.4 installed from the ROS Kinetic apt package `ros-kinetic-plotjuggler`.

For example, I create a `.csv` file in Pandas using the following commands:
```
import pandas as pd
import matplotlib as plt
df = pd.DataFrame(np.random.randn(6,4), index=[1,2,3,4,5,6], columns=list('ABCD'))
df.to_csv('test.csv')
```
the output is [this file](https://github.com/facontidavide/PlotJuggler/files/1083204/test.txt) (the extension has been changed to `.txt` because GitHub does not support attaching `.csv` files to issues). When loading the data into PlotJuggler, I get the following warning:
```
Selected time in not strictly  monotonic. Do you want to abort?
(Clicking "NO" you continue loading)
```
When clicking `No` I can see the data in the left pane, however, when trying to plot the data against the index vector, i.e., right-dragging the index vector to the plot pane and then left-dragging the data columns, this is the result I get:
[image]
Btw. I did maximize both the vertical and the horizontal zoom. When maximizing the vertical zoom, the following gets printed to the console from which I started PlotJuggler:
```
invalid X0/X1 range in PlotWidget::maximumRangeY
invalid X0/X1 range in PlotWidget::maximumRangeY
invalid X0/X1 range in PlotWidget::maximumRangeY
invalid X0/X1 range in PlotWidget::maximumRangeY
```

I expect to see this plot:
[image]
which is what I get by:
```
df.plot()
plt.show()
```

Am I missing something obvious?

facontidavide, 2017-06-19:
I will take a look at it. 

Cheers

facontidavide, 2017-06-19:
Fixed. There was a bug and also a small issue related to the fact that your first column doesn't have a name. 
I modified the code to insert a name like _Column_# when needed

closed 2017-06-19 by facontidavide

dmiklic, 2017-06-19:
Wow, that was quick! Thanks a lot, problem solved :)