PlotJuggler/PlotJuggler #47: Rule renaming issue

gstavrinos, 2017-06-19:
I am trying to create a ruleset for my package. Everything is working as expected, except for the following rule:

 <rule pattern="nodes.#/resident_memory_percentage" alias="nodes.#/name" substitution="@/resident_memory_percentage" />

If I use the rule above, I get a malloc() exception. If I use the rule below instead, everything is working perfectly:

 <rule pattern="nodes.#/resident_memory_percentage" alias="nodes.#/name" substitution="@/resident_memory_percen" />

The difference between the two is the string length of the subsctitution field. The crash happens in "resident_memory_percentage", "resident_memory_percentag", "resident_memory_percenta", "resident_memory_percent", and stops crashing at "resident_memory_percen".

The problem most probably lies in the ros_type_introspection package(in renamer.cpp at the "applyNameTransform" method?), but since it was encountered while using plotjuggler, I posted the issue here.

facontidavide, 2017-06-19:
Interesting. Would you kindly send me a rosbag file to recreate the issue ?

gstavrinos, 2017-06-19:
Please try with this rosbag. It's only a few seconds long, but it should be sufficient. Additionally, I am now running the latest version on my laptop, and without any rules I get a segmentation fault when I try to stream the same message. (rostune/node_stats)

Thanks for your time!

[2017-06-19-20-22-28.bag.zip](https://github.com/facontidavide/PlotJuggler/files/1085769/2017-06-19-20-22-28.bag.zip)

facontidavide, 2017-06-19:
I MIGHT have found the issue. This is the fix

https://github.com/facontidavide/ros_type_introspection/tree/fix

I will check this tomorrow to see if it works

facontidavide, 2017-06-20:
I confirm that the issue is fixed.
the issue was located in ros_type_introspection and you should update it.

Please check it out and close the issue if the issue is solved

gstavrinos, 2017-06-20:
After some git pulls I managed to compile both plotjuggler and ros_type_introspection, but I now get a segmentation fault when I try to stream the same topic. I ran plotjuggler with gdb using the following command `rosrun --prefix 'gdb -ex run' plotjuggler PlotJuggler`, and I get this error: (with or without rules)

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe3a9e9ca in std::vector<RosIntrospection::ROSMessage, std::allocator<RosIntrospection::ROSMessage> >::begin() const ()
   from /home/gstavrinos/catkin_ws/devel/lib/libros_type_introspection.so

facontidavide, 2017-06-20:
Hi,

I am not able to recreate the issue on my machine :(

To help me you should:

1) send me the message definitions
2) recompile the project in debug mode and see if gdb provides more information.

gstavrinos, 2017-06-20:
You can get the message definitions from here https://github.com/roboskel/rostune/tree/devel. (Notice the devel branch)

I ran plotjuggler again after compiling in debug mode, but got no additional information. Just that vague message that I posted earlier.

facontidavide, 2017-06-20:
I am deeply sorry, but it works just fine on my machine, I don't know how to help you.
By the way, you didn't send me a the backtrace of gdb (command __bt__).

gstavrinos, 2017-06-20:
You are right. I am sorry. This is the backtrace:

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe3c98a4e in std::vector<RosIntrospection::ROSMessage, std::allocator<RosIntrospection::ROSMessage> >::begin (this=0x0)
    at /usr/include/c++/4.8/bits/stl_vector.h:548
548       { return const_iterator(this->_M_impl._M_start); }
(gdb) bt
#0  0x00007fffe3c98a4e in std::vector<RosIntrospection::ROSMessage, std::allocator<RosIntrospection::ROSMessage> >::begin (this=0x0)
    at /usr/include/c++/4.8/bits/stl_vector.h:548
#1  0x00007fffe39da8c9 in RosIntrospection::__lambda2::operator() (
    __closure=0xd59de0, tree_node=...)
    at /home/gstavrinos/catkin_ws/src/ros_type_introspection/src/deserializer.cpp:94
#2  0x00007fffe39dbdf8 in std::_Function_handler<void(RosIntrospection::StringTreeLeaf), RosIntrospection::buildRosFlatTypeImpl(const ROSTypeList&, const RosIntrospection::ROSType&, RosIntrospection::StringTreeLeaf, uint8_t**, RosIntrospection::ROSTypeFlat*, uint16_t)::__lambda2>::_M_invoke(const std::_Any_data &, RosIntrospection::StringTreeLeaf) (__functor=..., __args#0=...)
    at /usr/include/c++/4.8/functional:2071
#3  0x00007fffe39dcd3b in std::function<void (RosIntrospection::StringTreeLeaf)>::operator()(RosIntrospection::StringTreeLeaf) const (this=0x7fffffffb1d0, 
    __args#0=...) at /usr/include/c++/4.8/functional:2471
#4  0x00007fffe39db240 in RosIntrospection::buildRosFlatTypeImpl (type_list=..., 
    type=..., tree_node=..., buffer_ptr=0x7fffffffb250, 
    flat_container=0x7fffe0963760 <DataStreamROS::topicCallback(boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::string const&)::flat_container>, 
    max_array_size=1000)
    at /home/gstavrinos/catkin_ws/src/ros_type_introspection/src/deserializer.cpp:170
#5  0x00007fffe39db3ee in RosIntrospection::buildRosFlatType (type_map=..., 
    type=..., prefix=..., buffer_ptr=0xd59c30 "\020\003", 
    flat_container_output=0x7fffe0963760 <DataStreamROS::topicCallback(boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::string const&)::flat_container>
, max_array_size=1000)
    at /home/gstavrinos/catkin_ws/src/ros_type_introspection/src/deserializer.cpp:203
#6  0x00007fffe06e11cb in DataStreamROS::topicCallback (this=0xc0ef70, msg=..., 
    topic_name=...)
    at /home/gstavrinos/catkin_ws/src/PlotJuggler/plugins/ROS/DataStreamROS/datastream_ROS.cpp:70
#7  0x00007fffe070c6a4 in boost::_mfi::mf2<void, DataStreamROS, boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::string const&>::operator() (
    this=0xd518c0, p=0xc0ef70, a1=..., a2=...)
    at /usr/include/boost/bind/mem_fn_template.hpp:280
#8  0x00007fffe0709979 in boost::_bi::list3<boost::_bi::value<DataStreamROS*>, boost::arg<1>, boost::_bi::value<std::string> >::operator()<boost::_mfi::mf2<void, DataStreamROS, boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::string const&>, boost::_bi::list1<boost::shared_ptr<topic_tools::ShapeShifter const> const&> > (this=0xd518d0, f=..., a=...) at /usr/include/boost/bind/bind.hpp:392
#9  0x00007fffe0706d16 in boost::_bi::bind_t<void, boost::_mfi::mf2<void, DataStreamROS, boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::string const&>, boost::_bi::list3<boost::_bi::value<DataStreamROS*>, boost::arg<1>, boost::_bi::value<std::string> > >::operator()<boost::shared_ptr<topic_tools::ShapeShifter const> > (this=0xd518c0, a1=...) at /usr/include/boost/bind/bind_template.hpp:47
#10 0x00007fffe0703549 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf2<void, DataStreamROS, boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::string const&>, boost::_bi::list3<boost::_bi::value<DataStreamROS*>, boost::arg<1>, boost::_bi::value<std::string> > >, void, boost::shared_ptr<topic_tools::ShapeShifter const> const&>::invoke (
    function_obj_ptr=..., a0=...)

    at /usr/include/boost/function/function_template.hpp:153
#11 0x00007fffe070caab in boost::function1<void, boost::shared_ptr<topic_tools::ShapeShifter const> const&>::operator() (this=0xd51780, a0=...)
    at /usr/include/boost/function/function_template.hpp:767
#12 0x00007fffe0709c48 in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<topic_tools::ShapeShifter const> const&)>, void, boost::shared_ptr<topic_tools::ShapeShifter const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<topic_tools::ShapeShifter const>) (
    function_obj_ptr=..., a0=...)
    at /usr/include/boost/function/function_template.hpp:153
#13 0x00007fffe071aa4a in boost::function1<void, boost::shared_ptr<topic_tools::ShapeShifter const> >::operator() (this=0xd51688, a0=...)
    at /usr/include/boost/function/function_template.hpp:767
#14 0x00007fffe0719528 in ros::SubscriptionCallbackHelperT<boost::shared_ptr<topic_tools::ShapeShifter const> const&, void>::call (this=0xd51680, params=...)
    at /opt/ros/indigo/include/ros/subscription_callback_helper.h:144
#15 0x00007fffe34e4645 in ros::SubscriptionQueue::call() ()
   from /opt/ros/indigo/lib/libroscpp.so
#16 0x00007fffe349e107 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
    () from /opt/ros/indigo/lib/libroscpp.so
#17 0x00007fffe349ec53 in ros::CallbackQueue::callAvailable(ros::WallDuration) ()
   from /opt/ros/indigo/lib/libroscpp.so
#18 0x00007fffe06e1d9e in DataStreamROS::extractInitialSamples (this=0xc0ef70)
[... cut ...]
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#51 0x00007ffff79a10e1 in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#52 0x00000000004fa430 in main (argc=1, argv=0x7fffffffd828)
    at /home/gstavrinos/catkin_ws/src/PlotJuggler/plotter_gui/main.cpp:124

I have tested it with two different Ubuntu 14 x64 machines, and I get a segmentation fault on both of them. Maybe there's a problem with my setup, since I try to have almost identical setups on all of my machines.

facontidavide, 2017-06-20:
fixed

closed 2017-06-20 by facontidavide