PlotJuggler/PlotJuggler #49: Problem with repeated values on x-axis

dmiklic, 2017-06-19:
A repeated value on the x-axis inhibits further plotting. [This file](https://github.com/facontidavide/PlotJuggler/files/1085785/test.txt) demonstrates the issue. Upon loading the file, if I select `_Column_0` as the time axis, which has identical values in rows 3 and 4, when I try to add e.g. column A to the plot area, only the first two values of A will be plotted.

My actual use case is a dataset of densely sampled (1ms) data. Occasionally, I will have two data points with the same timestamp, which is not really an issue, however, it seems to prevent PlotJuggler from correctly plotting the data.

This is with the currrent master head.

facontidavide, 2017-06-19:
I am afraid that this time I will not be able to give you a quick fix.. 
The assumption of monotonic time is made in many parts of the code,being a notable example the publisher plugin.

dmiklic, 2017-06-19:
I guess I got a bit greedy, seeing how quickly you made the other fixes. I suppose I can put some effort into cleaning my data up a bit :)

facontidavide, 2017-06-19:
it is really a problem of data structures. It might look as a simple feature form the outside but it will affect many part of the code, not just the XY plot.
in other words, I am afraid I will prioritize other features, since this one require a lot of work for a comparably small gain.

Don't hesitate anyway to ask for improvements. if I can, I am usually happy to work on them.

dmiklic, 2017-06-19:
Sure I understand. Anyways, 2/3 fixes within 24 hours is a very decent score :)

dmiklic, 2017-06-21:
Could we leave the issue open? I understand that It won't get fixed in the near future (or maybe ever :) and there are straightforward workarounds, but I feel like the feature itself is not unreasonable.

facontidavide, 2017-06-21:
of course it is not unreasonable ;)

facontidavide, 2018-08-16:
One year later...

closed 2018-08-16 by facontidavide

dmiklic, 2018-08-20:
Great, thanks! :)