PlotJuggler/PlotJuggler #50: Renaming rules problem persists

gstavrinos, 2017-06-21:
Ok, I tried in a completely different machine and setup, and plotjuggler is now working without any additional rules, but when I add either of the following rules, it crashes:

- <rule pattern="nodes.#/resident_memory" alias="nodes.#/name" substitution="@/resident_memory" />
-  <rule pattern="nodes.#/resident_memory_percentage" alias="nodes.#/name" substitution="@/resident_memory_percentage" />
-  <rule pattern="nodes.#/all_memory_percentage" alias="nodes.#/name" substitution="@/all_memory_percentage" />
-  <rule pattern="nodes.#/cputime_percentage" alias="nodes.#/name" substitution="@/cputime_percentage" />

Again, if I, for example, change "resident_memory" to "resident_mem", everything is working perfectly!

facontidavide, 2017-06-21:
Are you using ros_type_introspection version 0.6.0 ?

gstavrinos, 2017-06-21:
Ι downloaded a fresh clone from the master branch, earlier today.

facontidavide, 2017-06-22:
just pushed a potential fix to ros_type_introspection. Take a look

gstavrinos, 2017-06-22:
Yes!!! It's working!! Thank you very much for your time and dedication! (You can still consider renaming plotjuggler to harryplotter!)

closed 2017-06-22 by gstavrinos