PlotJuggler/PlotJuggler #53: Stream in a non ros-based environment

JimmyDaSilva, 2017-09-13:
Hello @facontidavide !

First thank you for your big contribution to the ROS community. PlotJuggler is really easy to use and as already saved me a lot of debug and figure creation time.

I am now unfortunately working on a non ros-based project on Windows, so I cannot use PlotJuggler for streaming any more...

Could it be possible somehow to include PlotJuggler as a new Widget inside my QT based application to stream some variables ?

Jimmy

facontidavide, 2017-09-13:
Hi @JimmyDaSilva ,

First of all, you should be able to compile PlotJuggler in Windows WITHOUT the ROS plugin. If you can't let me know the compilation error.

About including  PlotJuggler as a widget, It MIGHT be possible, but it is not on my todo list right now.

Anyway, you will need to build your own streaming plugin. it is relative easy, you might want to take a look to this example: https://github.com/facontidavide/PlotJuggler/blob/master/plugins/DataStreamSample/datastream_sample.cpp

if you manage to laucnh PlotJuggler inside another Qt project, you might use slot-signals to pass data. otherwise, if you want to pass data from a certain producer to PlotJuggler, you should use an inter-process communication. I usually suggest people to use ZeroMQ + Protobuffer, since they are easy to learn.

JimmyDaSilva, 2017-09-13:
@facontidavide , the compilation under Windows did work like a charm.
I will have a look at the example to see if I can work out something about it easily.

Thanks :D

facontidavide, 2017-09-13:
With your permission, I will close the issue, since there isn't much I can do.

If you work for a company and you need support to implement a custom plugin, I can do it for you as a freelancer. 
If the specific plugin can help a broader community and it is not tailored to your specific use case, I will be happy to do it for free ;)

closed 2017-09-13 by facontidavide

JimmyDaSilva, 2017-09-13:
Sure you can close.

I am working in a public robotics lab based in Paris, called ISIR.

Being able to use this as a qtwidget in any qt based gui would be for sure helpful for many people.

facontidavide, 2017-10-16:
Awesome

JimmyDaSilva, 2018-01-05:
See my collegue's pull request #64