PlotJuggler/PlotJuggler #54: Easy reload of files

SammyRamone, 2017-09-22:
Hi,

just an enhancement request to have a function for reloading the same data file.
I am currently working with a .csv data file which I generate with some code and then visualize with PlotJuggler.
Now every time I generate a new data set I have to click on "Load data from: x.csv" and reselect the time axis index. It would be awesome to just have a button (or even a shortcut like F5) to reload the last file with the same time axis.
But as I said, it's just an enhancement and it already works quiet well.

Thanks again for your amazing tool! It should be the standard plotter in ROS :smiley:

facontidavide, 2017-09-25:
sure, I will take a look at it during this week. cheers

facontidavide, 2017-11-19:
hi. I think I addressed the issue you mentioned.

It actually helped me to fix some bugs that I didn't know I had.

Let me know if the action "Files -> Quick Data Reload" is what you had in mind

closed 2017-11-19 by facontidavide

SammyRamone, 2017-11-23:
It works perfect, thanks a lot :)