PlotJuggler/PlotJuggler #56: Gray Subscriber label

AARobotics, 2017-10-15:
Hey, thank you for the tool.

The Subscriber is not enabled and I am unable to subscribe to any topic.

PlotJuggler version: 1.3
ros: Kinetic
Ubuntu 16.04

Thanks

facontidavide, 2017-10-15:
thank you for reporting, i will check it out as soon as possible

facontidavide, 2017-10-15:
I just checked the version on master and it seems to work just fine (at least on Indigo). Can you send me a video or screenshot to see the problem?

In the console, do you see this message at startup?

       "libDataStreamROS.so" : is a DataStreamer plugin

AARobotics, 2017-10-16:
Hey thanks for the fast response.

Terminal output:
$ rosrun plotjuggler PlotJuggler 
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.

screenshot:

[image]

facontidavide, 2017-10-16:
Mistery solved. Are you compiling with catkin? If you use cmake without catkin, all the ROS plugins will be ignored

AARobotics, 2017-10-16:
Hey actually I tried using apt-get install ros-kinetic-plotjuggler
when it didnt work I deleted it and the installed it directly in the src folder of the catkin workspace. Then installed it as instructed (cmake) and then catkin_make. Still same issue. Maybe its due to the kinetic version?

facontidavide, 2017-10-16:
Ok, I will check the kinetic version

facontidavide, 2017-10-16:
Version 1.3 of PlotJuggler FAILS to compile if you don't hve the ltest version of ros_type_introspection installed (1.0.0).

In theory you should get a compilation error when you invoke catkin_make, if the ros_type_introspection doesn't match

AARobotics, 2017-10-16:
Hey
Thank you very much! it was the issue. apt-get installed ros_type_introspection 0.9
The solution is obviously to copy past the git following folders to the src:

https://github.com/facontidavide/ros_type_introspection
https://github.com/Eurecat/abseil-cpp

Thank you very much!! Looking forward using your tool.

closed 2017-10-16 by AARobotics

facontidavide, 2017-10-16:
with the next syn all the new version will be installed at once ;)

You are welcome, thank you for reporting. Don't hesitate to send an issue if you have more questions