PlotJuggler/PlotJuggler #59: "Segmentation fault (core dumped)" error

hassan82, 2017-11-14:
Hello,

I am eager to use your program. I have just installed 1.3.0 version of PlotJuggler using catkin_make on my Ubuntu14.04 ros-indigo installed machine. Firstly, I tried to use the dummy data with argument "-t" and it visualized them successfully.  When I record a rosbag file and tried visualize I had an error of  "Segmentation fault (core dumped)". 

I attached the rosbag file. I would be grateful if you could help me on my problem? 

Thanks in advance.

[2017-11-14-16-34-23.bag.zip](https://github.com/facontidavide/PlotJuggler/files/1470972/2017-11-14-16-34-23.bag.zip)

facontidavide, 2017-11-14:
Sure, I will take a look at it

facontidavide, 2017-11-14:
Thank you very much for reporting, you helped me found an error in another library of mine.

https://github.com/facontidavide/ros_type_introspection/commit/8c4fa7fb533fc81a8a2606fdb7d6d52a73e01f6d
The topic that is problematic is __/pedsim/static_obstacles__

You should compile the master branch of both __ros_type_introspection__ and __PlotJuggler__.
 I will subbit the changes to the ROS buildfarm but it will take a while before it is updated.

closed 2017-11-14 by facontidavide