PlotJuggler/PlotJuggler #60: 'DataLoad ROS Bags' throws exception when loading bag

airballking, 2017-11-15:
Hi,

I have encountered the following issue: When loading a bag the plugin 'DataLoad ROS Bags' throws an error. The specific text is "Buffer overrun in RosIntrospection::ReadFromBuffer". Here is a screenshot:

[image]

I am using PlotJuggler version 1.3.0 installed from debians using Ubuntu 14.04 and ROS Indigo. I tried compiling from source, but failed. Sorry.

This is the bag I tried to load:

[2017-11-15-09-33-32.bag.zip](https://github.com/facontidavide/PlotJuggler/files/1473918/2017-11-15-09-33-32.bag.zip)

I contains a single message of type ```control_msgs/FollowJointTrajectoryGoal``` with a trajectory with 500 points in it. I wanted to look at the position and velocity trajectories of individual joints using PlotJuggler. Is that usage in principal supported?

Best,
Georg.

facontidavide, 2017-11-15:
This was solved yesterday #59
Please use the version on master of both ros_type_introspection (1.0.1) and PlotJuggler (1.4.0)

Cheers

closed 2017-11-15 by facontidavide

airballking, 2017-11-15:
That's an interesting coincidence. I will try again with both ros_type_introspection and PlotJuggler compiled from source.

facontidavide, 2017-11-15:
Not a coincidence you and someone else reported it. I introduced a bug recently and people started noticing it :(