PlotJuggler/PlotJuggler #61: Loading valid .csv results in "Selected time in notstrictly  monotonic" error

mactro, 2017-11-20:
The first time I load a csv file to PlotJuggler there is a pop-up asking to select the time axis. The file is loaded correctly. When I want to load another file with same data structure, there is "Selected time in notstrictly  monotonic" error. 

The only way I can get data to be loaded correctly then, is to change the name of time axis in the csv.

Example files (extensions changed to txt due to Github restrictions):
[test.txt](https://github.com/facontidavide/PlotJuggler/files/1487087/test.txt)
[test2.txt](https://github.com/facontidavide/PlotJuggler/files/1487088/test2.txt)

facontidavide, 2017-11-20:
I think I fixed it... https://github.com/facontidavide/PlotJuggler/commit/48c5d0537257543650a03c1c16fe3d6703cd1a3b

facontidavide, 2017-11-20:
By the way, thank you very much for reporting

facontidavide, 2017-11-20:
Not quite there... I introduced an error :P

facontidavide, 2017-11-20:
ok. know it works

closed 2017-11-20 by facontidavide