PlotJuggler/PlotJuggler #67: Difficulty compiling plotjuggler

ahundt, 2018-01-26:
I'm trying to install plotjuggler for the first time and I'm encountering issues similar to #56, it seems some of the dependencies in the readme aren't up to date. Specifically,  

- an error for missing qt web sockets
- a warning for messing benchmark (what repository is this from exactly?)

Here are the steps I took, I'm on kinetic:
```
cd myworkspace/src
sudo apt-get -y install qtbase5-dev libqt5svg5-dev
git clone https://github.com/facontidavide/PlotJuggler.git
git clone https://github.com/Eurecat/abseil-cpp
git clone https://github.com/facontidavide/ros_type_introspection
catkin build plotjuggler
```

This gives me the following error:
```
Warnings   << ros_type_introspection:make /home/ahundt/src/costar_ws/logs/ros_type_introspection/build.make.000.log                                                                                                                                                                    
/home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp: In member function ‘void RosIntrospection::Parser::registerRenamingRules(const RosIntrospection::ROSType&, const std::vector<RosIntrospection::SubstitutionRule>&)’:
/home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp:141:47: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
     if( rule_set.find( rule ) == rule_set.end() )
                                               ^
In file included from /usr/include/c++/5/bits/hashtable.h:35:0,
                 from /usr/include/c++/5/unordered_set:47,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_introspection.hpp:38,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp:42:
/usr/include/c++/5/bits/hashtable_policy.h:317:5: note: candidate 1: bool std::__detail::operator==(const std::__detail::_Node_iterator_base<_Value, _Cache_hash_code>&, const std::__detail::_Node_iterator_base<_Value, _Cache_hash_code>&) [with _Value = RosIntrospection::SubstitutionRule; bool _Cache_hash_code = true]
     operator==(const _Node_iterator_base<_Value, _Cache_hash_code>& __x,
     ^
In file included from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_type.hpp:42:0,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_field.hpp:42,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_message.hpp:39,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/stringtree_leaf.hpp:43,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_introspection.hpp:39,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp:42:
/home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/utils/variant.hpp:140:6: note: candidate 2: bool RosIntrospection::operator==(const T&, const RosIntrospection::Variant&) [with T = std::__detail::_Node_iterator<RosIntrospection::SubstitutionRule, true, true>]
 bool operator ==(const T& num, const Variant& var)
      ^
/home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp:141:47: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
     if( rule_set.find( rule ) == rule_set.end() )
                                               ^
In file included from /usr/include/c++/5/bits/hashtable.h:35:0,
                 from /usr/include/c++/5/unordered_set:47,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_introspection.hpp:38,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp:42:
/usr/include/c++/5/bits/hashtable_policy.h:317:5: note: candidate 1: bool std::__detail::operator==(const std::__detail::_Node_iterator_base<_Value, _Cache_hash_code>&, const std::__detail::_Node_iterator_base<_Value, _Cache_hash_code>&) [with _Value = RosIntrospection::SubstitutionRule; bool _Cache_hash_code = true]
     operator==(const _Node_iterator_base<_Value, _Cache_hash_code>& __x,
     ^
In file included from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_type.hpp:42:0,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_field.hpp:42,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_message.hpp:39,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/stringtree_leaf.hpp:43,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/ros_introspection.hpp:39,
                 from /home/ahundt/src/costar_ws/src/ros_type_introspection/src/ros_introspection.cpp:42:
/home/ahundt/src/costar_ws/src/ros_type_introspection/include/ros_type_introspection/utils/variant.hpp:134:6: note: candidate 2: bool RosIntrospection::operator==(const RosIntrospection::Variant&, const T&) [with T = std::__detail::_Node_iterator<RosIntrospection::SubstitutionRule, true, true>]
 bool operator ==(const Variant& var, const T& num)
      ^
cd /home/ahundt/src/costar_ws/build/ros_type_introspection; catkin build --get-env ros_type_introspection | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.......................................................................................................................................................................................................................................................................................
Finished  <<< ros_type_introspection                [ 9.8 seconds ]                                                                                                                                                                                                                    
Starting  >>> plotjuggler                                                                                                                                                                                                                                                              
_______________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << plotjuggler:cmake /home/ahundt/src/costar_ws/logs/plotjuggler/build.cmake.001.log                                                                                                                                                                                        
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:26 (find_package):
  Could not find a package configuration file provided by "Qt5WebSockets"
  with any of the following names:

    Qt5WebSocketsConfig.cmake
    qt5websockets-config.cmake

  Add the installation prefix of "Qt5WebSockets" to CMAKE_PREFIX_PATH or set
  "Qt5WebSockets_DIR" to a directory containing one of the above files.  If
  "Qt5WebSockets" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  plugins/DataStreamServer/CMakeLists.txt:8 (find_package)

cd /home/ahundt/src/costar_ws/build/plotjuggler; catkin build --get-env plotjuggler | catkin env -si  /usr/local/bin/cmake /home/ahundt/src/costar_ws/src/PlotJuggler --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/ahundt/src/costar_ws/devel/.private/plotjuggler -DCMAKE_INSTALL_PREFIX=/home/ahundt/src/costar_ws/install; cd -
.......................................................................................................................................................................................................................................................................................
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                                                                                                
[build] Runtime: 16.8 seconds total.                                                                                                                                                                                                                                                   
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

cc: @Chunting

facontidavide, 2018-01-28:
Thanks for reporting. I was kind of aware of this issue, but a bit too lazy to fix it promptly ;)

My last commit should fix it.

closed 2018-01-28 by facontidavide

ahundt, 2018-01-29:
Cool thanks, that worked!