PlotJuggler/PlotJuggler #68: Can PlotJuggler help with plotting data indexed from multiple clocks?

ahundt, 2018-01-26:
I have data indexed with multiple time samples from 3 separate clocks across 3 separate devices, and I'm trying to visualize the time differences and errors, I've included my specific questions and details below. 

1. Do you think plotjuggler is the right tool for such a use case?
2. Which of the following would be easiest step to view my data? It is currently logged to disk in a [flatbuffers](github.com/google/flatbuffers) message (like a rosmsg or protobuf). Options include:
    1. dump the data into a rosbag
    2. dump to CSV (must every csv entry be occupied or can some be blank?)
    3. write a custom plugin
    4. another alternative you might advise

The key data channels will be:
- A 3D camera's clock time
    - 6DOF poses of objects the camera can see
- A robot arm's clock time
   - 7DOF joint angles, and possibly torque
   - 6DOF end effector pose estimate
- The computer clock time immediately before and after receiving the above two times.

The actual message data type [LogKUKAiiwaFusionTrack.fbs](https://github.com/ahundt/grl/blob/master/include/grl/flatbuffer/LogKUKAiiwaFusionTrack.fbs) is simply a vector of time stamped messages, which contain other data channels:
```
union DeviceState {KUKAiiwaState, FusionTrackMessage}
// a log file is a series of GRLMessage entities
// The TimeEvent should contain the necessary data,
// and only ONE of kukaiiwa or fusionTrack should
// be filled out for each message.
table KUKAiiwaFusionTrackMessage {
    // an easy to use ros compatible timestamp
    timestamp:double;
    // a more detailed timestamp based on google cartographer
    // and the triggersync paper
    timeEvent:TimeEvent;
    deviceState:DeviceState;
}

table LogKUKAiiwaFusionTrack {
  states:[KUKAiiwaFusionTrackMessage];
}
```

Here is the time data I record for each device:
```
/// Note that all of these time entries are
/// longs with a minimum step of 100ns,
/// see google cartographer's cartographer::common::time
table TimeEvent {
  /// Identifying string for this time stamped data topic
  /// something like "/opticaltracker/00000000/frame" where
  /// 00000000 is the serial number of the optical tracker.
  event_name:string;
  /// The time just before a data update request is made
  local_request_time:long;
  /// Identifying string for the clock used to drive the device
  /// something like "/opticaltracker/00000000/clock"
  /// if it is the clock internal to a sensor like an optical tracker
  device_clock_id:string;
  /// The time provided by the device specified by device_clock_id
  device_time:long;
  /// Identifying string for the clock used to drive the device
  /// or "/control_computer/clock/steady" if the device has no clock
  /// and the time is the desktop computer
  /// running the steady clock (vs clocks which might change time)
  local_clock_id:string;
  /// The time at which the data was received
  local_receive_time:long;
  /// The corrected local time which represents when the sensor
  /// data was actually captured.
  corrected_local_time:long;
  /// Estimated duration of the skew between the device clock
  /// and the local time clock
  clock_skew:long;
  /// The minimum expected delay in transporting the data request
  min_transport_delay:long;
}
```

This is based on the rosmsg [Event.msg](https://github.com/englishar/trigger_sync/blob/master/msg/Event.msg), from a paper called [triggersync](https://wiki.qut.edu.au/download/attachments/147952130/TriggerSync.pdf.pdf?version=1&modificationDate=1426037989000&api=v2) as follows (there is a TimeStamped version as well): 

```
string  event_name
time    local_request_time
string  device_clock_id
time    device_time
string  local_clock_id
time    local_receive_time
time    corrected_local_time
float64 clock_skew
duration    min_transport_delay
```

I would appreciate your patience and recommendation, I know this is a very detailed question. Others in my lab use plotjuggler and it is highly regarded, so I'm seeing if I can use it too. Thanks!

ahundt, 2018-01-26:
cc @Chunting

facontidavide, 2018-01-28:
Let's see if I can help you...

1) dump the data into a rosbag
if you can do it easily, that might be a shortcut.   

2)  dump to CSV (must every csv entry be occupied or can some be blank?)   
Yes, every entry must be occupied :(

3) write a custom plugin
it is not terribly hard, but time consuming for sure. I can help you if you want.

4) another alternative you might advise (is there a way to load json?)
Nope. Options 1, 2, 3 are your best shot

Let me think about the other problem (comparing clocks) to see if I can figure out a solution :)

ahundt, 2018-01-29:
Cool, thanks! That gives me some direction. I can start with just CSV with a "base" clock + differences. There will be some discontinuities and the file might be very large because I'll be looking at ~3 minutes data recorded with 1-4ms timesteps, but it would get me started.

facontidavide, 2018-01-29:
If I correctly understood, you have to load a single log file, which contains all the information.

Looking at the flatbuffer description, few things come to my mind:

Keep in mind that a time series in PlotJuggler is identified by a name (a single string) and a sequence of [time,value] pairs of doubles.
There is no way to plot a string. I am sayind that because some of the fields of timeEvent are string, such as __device_clock_id, event_name and local_clock_id__. 

Therefore, I expect the name of the time serie to be a combination of device_clock_id, event_name and local_clock_id AND the field name you want to plot (for instance device_time).

Do not forget that you also need to specify the X value (time axis) of the series; it is not clear to me which one you want to use.

In an ideal world what do you think that the plot name, the X axis and Y axis of a plot should be?

ahundt, 2018-01-30:
The strings are basically a "rostopic" string so the different devices can be told apart. In a CSV that will be the title.

What I might be able to do something like the following pseudocode to write out the csv:

```
# on the first message save local_request_time as the initial_local_request_time
# on the first message save device_time as the initial_device_time
X = local_request_time
local_request_offset = (local_request_time - initial_local_request_time)
current_device_offset = (device_time - initial_device_time)
Y = current_device_offset - local_request_offset
```

Then Y should drift around based on how the clocks change relative to each other without any crazy units.

ahundt, 2018-02-03:
Ok we've been able to plot the data via CSV by plotting all numbers as offsets relative to a single clock, but we have run into a problem.
Simply merging the data from the two separate devices into one CSV by repeating previous values leads to the appearance that all the motion or no motion happened over very short periods of time, and thus there are massive discontinuities in the data.

For that reason, another approach is necessary. Is it possible to load two separate csv files and plot the graphs together? That has potential to do the trick.

facontidavide, 2018-02-05:
The problem is that  loading two files  with the same field names is not possible, in theory. 
To fix that, I just added a new feature to the Dialog that is showed when a rosbag is loaded.

I can add it also to the CSV plugin, that will take some more work

Chunting, 2018-02-05:
[image]
[image]

This is the time offset we ploted via CSV.

facontidavide, 2018-02-05:
is that... good news or bad news? I don't know what to think...
Is the  Y axis scale nanoseconds?

ahundt, 2018-02-06:
Unfortunately, that's bad news, the giant vertical jumps in the green bar are due to those discontinuities I was mentioning, rather than the actual data. Just trying to demonstrate the issue we're trying to resolve by loading two separate CSVs. We can give you sample CSVs if you'd like.

facontidavide, 2018-02-08:
This commit might help a little. https://github.com/facontidavide/PlotJuggler/commit/0f2d09beb924c6dd70182b617db5a3d9a7cd840f
 
Know it is possible to parse CSV files with empty cells

[image]

ahundt, 2018-02-10:
cool that might do the trick! we will give it a try

facontidavide, 2018-02-14:
Is there anything else I can do for you? Can we close this issue?

Chunting, 2018-02-14:
Thank you.
If some cell in the first line is empty,  PlotJuggler can’t recognize it and ignore it. I need to fill it out by zero manually, can you fix it? 
It can’t recognize FT_local_request_time, FT_device_time_offset and Y_FT.

Also could you please give more information about this error? “DataLoadCSV: problem parsing the first two lines”, sometimes I cannot plot data because of this error.

Thanks 
Chunting

ahundt, 2018-02-15:
Thank you for the update! It is mostly working but as @Chunting mentioned, there seems to be minor bugs remaining for this use case.

@Chunting  Would you mind actually uploading a CSV file to this github issue which demonstrates the problematic behaviors so he can actually reproduce and fix it? You can do it directly from [this issue's page on the github website](https://github.com/facontidavide/PlotJuggler/issues/68), since it appears you're using the email interface.

@facontidavide I wanted to thank you for this project it is turning out to work terrifically for us, especially with how easy it is to load up and see data at both a high level and at the finest levels of detail, plus it has a very responsive interface!

facontidavide, 2018-02-15:
@Chunting 
Also could you please give more information about this error? “DataLoadCSV: problem parsing the first two lines”, sometimes I cannot plot data because of this error.

"Luke
[image]

A sample file to reproduce the error would help

Chunting, 2018-02-15:
Thanks, there was a bug in my code and I fixed it.
But how about this problem?
If some cell in the first line is empty, PlotJuggler can’t recognize it or ignore it. I need to fill it out by zero manually, can you fix it? It can’t recognize kuka_local_request_time, kuka_device_time_offset and Y_kuka.
[image]
[image]

Thanks,
Chunting

facontidavide, 2018-02-15:
I just made a commit that might solve your issue.

please, if the problem persist, SEND ME the file. If it is solved, let me know

Chunting, 2018-02-15:
Here is the CSV file.
[2018_02_15_15_47_34_FTKUKA.csv.zip](https://github.com/facontidavide/PlotJuggler/files/1729755/2018_02_15_15_47_34_FTKUKA.csv.zip)

Thanks.

facontidavide, 2018-02-16:
With the last modifications I made yesterday, the file you sent loads just fine. close the issue if you can confirm on your side that it works

ahundt, 2018-02-17:
My apologies for all the trouble here, this is (to my knowledge) @Chunting's first time trying to provide all the information necessary to reproduce a bug to someone on a public project whom he hasn't met. It was tricky my first time too so I appreciate your patience!

I'll talk to him and see if there was still an issue he is able to reproduce or if that's it, I think there might have been when the CSV starts with blanks on the first line. 

Thanks a million for your help!

facontidavide, 2018-02-17:
@ahundt and @Chunting, no need to apolgize !
I am happy to help and I am glade that you are both helping suggesting improvements and reporting bugs! If people don't report this kind of issue, I can not make PlotJuggler better on my own.

But I just can't resist the temptation to use silly memes... like this one:

[image]

Chunting, 2018-02-18:
It works as well as expected. You can close this issue now. Thanks.

closed 2018-02-18 by facontidavide