PlotJuggler/PlotJuggler #69: PlotJuggler layout start via launch file

CesMak, 2018-02-02:
Hey there,

I wish, that PlogJuggler would be able to be started via launch files with a predefined layout. Unfortunately this is not possible actually if you start gazebo in paused mode (In case your predefined layout contains topics that are first published when gazebo is started). 

Currently it is like this: I Start plotjuggler with a predefined layout from a launch file then it says it cannot open the topics of this layout cause the topics are not yet published (because gazebo is started in paused mode)

That is why it would be better that plotjuggler just prints a message like: 

waiting for topics to come in.... (When loading a layout that has topics which are not published yet) 

Hope you can do this improvement on the code :)

facontidavide, 2018-02-02:
I will work on this. Thanks for reporting.

facontidavide, 2018-02-05:
I think I fixed your issue. Please check it out

facontidavide, 2018-02-05:
I realized that there is more work to do...

closed 2018-02-07 by facontidavide