PlotJuggler/PlotJuggler #70: Issue loading sensor_msgs/Image messages from a rosbag

alex-silva-fiveai, 2018-02-07:
Hi,

Thanks for the great work on PlotJuggler (great name btw).

I seem to have found an issue when loading `sensor_msgs/Image` messages from a rosbag.
I am currently using the latest master version.

Error message displayed:
```
The plugin [DataLoad ROS bags] thrown the following exception: 

 Buffer overrun in deserializeIntoFlatContainer (blob)
```
I am not the owner of the rosbag thus cannot share it unfortunately.
Let me know if there is any other piece of information that would help resolving this issue.

Alex

facontidavide, 2018-02-07:
Thanks for reporting. An exception in thrown from https://github.com/facontidavide/ros_type_introspection
I have to fix the problem there. Once it is done, you will have to use the master version of ros_type_introspection and compile PlotJuggler again.

I will let you know

facontidavide, 2018-02-08:
Fixed in ros_type_introspection 1.0.2 (current master). 
https://github.com/facontidavide/ros_type_introspection/releases/tag/1.0.2
You should recompile PlotJuggler using that version. I also uploaded a video that show how to use PlotJuggler and rqt_image_view at the same time.
https://vimeo.com/254831364

closed 2018-02-08 by facontidavide

alex-silva-fiveai, 2018-02-08:
Can confirm it works now. Thanks @facontidavide