PlotJuggler/PlotJuggler #71: Improvements: Saving data to CSV

JimmyDaSilva, 2018-02-07:
Hi @facontidavide, I think it would be a simple great improvement to just be able to save the data to a CSV file, in addition to saving as .png

Cheers,
Jimmy

facontidavide, 2018-02-07:
It sounds like a very nice feature to have. I will work on it in my spare time. Thanks for the suggestion

romainreignier, 2019-02-17:
I also find that this feature could be useful but I am not sure if it is best to only save the data in the current plot (closer to the current "Save plot to file" option) or save the all the loaded timeseries, including the custom ones, or the one filtered by the top left filter?

I ask this question here to help anyone wanting to implement such feature.

facontidavide, 2019-02-17:
Don't keep your breath waiting for this feature, since it is at the bottom of my priorities right now...

Unless someone wants to sponsor or contribute with code to this feature.

The main reason is that it is not as trivial as it looks like since CSV works best when all the series share the same time axis (same amount of points).

This is not the case in general. There are solutions of course, but it takes an amount of time that I prefer to work on other features.

romainreignier, 2019-02-17:
Ok, I see the point.

Thanks for the details.

peci1, 2020-10-29:
Maybe the approach taken in https://github.com/AtsushiSakai/rosbag_to_csv could be used - save each topic to its own CSV. That way, solving timestamp issues would be left to the user.

griffindore, 2021-05-13:
csv with 3 columns: time, attribute_name, value

Load that bad-boy in any DataFrame package of your choice and do whatever you want.

griffindore, 2021-05-13:
@facontidavide you don't have to boil the ocean. I did not underestimate anything. I proposed that you don't try to make a sparse matrix. You don't need to create a column for every attribute in the plot. Make 3 columns: attribute_name, value, and timestamp. Every row is then a single sample of a single attribute.

Then a user can groupby the attribute they want, resample as they wish and combine attributes into a single table. I can do all of this in python and pandas in 5 minutes.

facontidavide, 2021-05-13:
That of course vould be easier. and useful for some users. But I would like to create CSV files that PlotJuggler itself can open.

Anyway, good timing, because I am currently working on this (a sponsor is paying for this feature).

Stay tuned...

facontidavide, 2021-09-19:
Done

closed 2021-09-19 by facontidavide