PlotJuggler/PlotJuggler #75: Behaviour when trying to plot a ROS message without stamp

SammyRamone, 2018-02-14:
Hello,

the behaviour of PlotJuggler is strange, when you try to plot a ROS topic where the messages have no timestamp in the header. Basically it only reads one message and displays it.
It took me a while to figure out what the problem is. Obviously there should be a stamp in the message, but maybe PlotJuggler could just display an error message to make it clearer to the user.

Here is a picture how it looks like.

[image]

I tested it on Ubuntu 16.04 with ROS Kinetic using the current master branch.

I can also send you a rosbag, but I guess it's the same with any stamp less message.
I'm glad to help, if you need any further information.

facontidavide, 2018-02-14:
Will this do?

https://github.com/facontidavide/PlotJuggler/commit/3b50856bb6ca56cce21cb1adc0bc80463883c9b8

closed 2018-02-14 by facontidavide

facontidavide, 2018-02-14:
I decided to add one more improvement. 

https://github.com/facontidavide/PlotJuggler/commit/cc06d702e672765fe69b529a79726e0fc27af548

Even if the checkbox is checked, if the stamp is exactly equal to 0 it is very probable that the user want the timestamp to be filled automatically now.