PlotJuggler/PlotJuggler #78: axis is stick to [-0.1, 0.1]

congwu26, 2018-03-20:
I used the streaming function, when I add the pose.x to the plot the y axis is somehow stick to [-0.1, 0.1] which I can not zoom in and out.

facontidavide, 2018-04-05:
With so little information, it is hard to know what is going on. 
When the Y axis is not updated, most probably you are not receiving any message or all the values are 0.
Please provide more information:

1) Are you streaming using the ROS plugin or your own?
2) If it is ROS, can you share a rosbag and a __layout file__ to reproduce this behavior?

v4hn, 2018-04-24:
Hey there,

I see the same problem with a particular `std_msgs/Float64` stream with the latest version available in kinetic over here.

Here are the files you requested. Notice that when the command value jumps back to 0 suddenly the limits are ok for a while... (until the jump leaves the observation window again).

[plotjuggler-stuck-limits.zip](https://github.com/facontidavide/PlotJuggler/files/1942214/plotjuggler-stuck-limits.zip)

facontidavide, 2018-04-24:
I will take a look.

closed 2018-04-24 by facontidavide

v4hn, 2018-04-24:
Wow, impressive response time. :-)

facontidavide, 2018-04-24:
You are welcome, I would have solved this sooner if @congwu26 shared his rosbag ;)

commit 25692d97 (2018-04-24): bug fix #78