PlotJuggler/PlotJuggler #79: Exact navigation to a data point

llvs, 2018-03-22:
Dear Davide,

I like PlotJuggler but I want to show you an issue I have with it: I want to find the value of the minimum peak in a rosbag file. In the screenshot you see that the red hairline is just right from the minimum. Using the time step slider and the arrow keys, one single press to the left moves the red hairline to the left side of the minimum and I seem unable to obtain the value that is exactly at the minimum position.

Is it possible to somehow drag the red hairline to obtain a finer time sampling than with the bottom slider?

[image]

In private communication I could also send you the corresponding bag file.

rosbag info gives:
```
version:     2.0
duration:    2:45s (165s)
start:       Mar 21 2018 17:00:30.33 (1521648030.33)
end:         Mar 21 2018 17:03:16.17 (1521648196.17)
size:        184.7 KB
messages:    2483
compression: none [1/1 chunks]
types:       XXXXX/timediff [10ec55469cc6072f685fdf8811f72155]
topics:      /timediff   2483 msgs    : XXXXX/timediff
```

facontidavide, 2018-03-22:
Hi,

the best way in my opinion to move the time slider is using the **[SHIFT + LEFT MOUSE]** combination on the plotting area.
In your case I would also zoom on the spike and activate the "show lines and points" option right clicking on the plot.
If you send me the rosbag at my email davide.faconti@gmail.com, I will do a video showing you how to do it.

Cheers

llvs, 2018-03-22:
OK, that helps. Thanks a lot!

closed 2018-03-22 by llvs