PlotJuggler/PlotJuggler #82: plot update missing when streaming off and switch to different tab

v4hn, 2018-05-01:
Hey again,

I use the latest master and build from source.

Suppose I stream data from ROS topics.
I create two tabs (tab 1 and tab 2) and add curves to the plots of each tab.
Now I observe the contents of tab 1 for a while, see an interesting behavior, and stop streaming via the "Streaming ON" button on the bottom left.
When I switch back tab 2, I expect to see the streamed data up to the moment I stopped it.
But this is not the case. The plot on tab 2 will still show old data instead.

If I do all of that rather quickly and continue streaming from tab 2 right away, the plot is updated again and I can see the correct data as expected.

facontidavide, 2018-05-01:
thanks for reporting, I will check this out

facontidavide, 2018-05-01:
Fixed! nice catch, thanks for reporting.

closed 2018-05-01 by facontidavide

v4hn, 2018-05-01:
As multiple times before: Thank you so much for all your work here!