PlotJuggler/PlotJuggler #85: ROS Kinetic Missing Plugin Libraries

evanglasgow, 2018-05-19:
Hello!

Upon updating to the most recent version of plotjuggler provided by the ROS APT system (`1.6.0-0xenial-20180501-141149-0800`), I noticed an issue where the plugin libraries appear to be missing. The system location `/opt/ros/kinetic/lib/plotjuggler/` only contains the the PlotJuggler application and none of the lib*.so plugins. So when attempting to open a bag file an error message appears that no plugin is available for that operation.

facontidavide, 2018-05-19:
Hi,

I will check it out, thanks for reporting

evanglasgow, 2018-05-20:
Thanks for looking into this so quickly! I'll keep an eye out for the next ROS package sync and compile it manually until the new release is available.

wangxiuzhong, 2018-05-26:
Hello!
I also encountered the same problem,could you tell me how compile it ?
I have tried to compile it by setting (COMPILING_WITH_CATKIN 1) in _CmakeLists.txt_ file,and it does not  work.The results in my console like this:
[image]

Thanks!

facontidavide, 2018-05-26:
Hi,

**you must NOT set COMPILING_WITH_CATKIN manually.**

That variable is set __automatically__  if you are compiling using catkin as you would do with any other ROS package, i.e. inside a catkin workspace.

You can use either catkin_make or catkin build.

wangxiuzhong, 2018-05-26:
Thank you for your reply !
I already use catkin_make in my workspace.However, no executable file was generated:
[image]

Sorry to bother you again!

facontidavide, 2018-05-26:
No problem, i am here to help!

I found it. The problem is in the WHITELIST.

The name of the package is **plotjuggler** (lower characters) not **PlotJuggler**

wangxiuzhong, 2018-05-26:
It works! Thank you very much for your help! :)

gavanderhoorn, 2018-05-30:
@facontidavide wrote:
> I think I know what happened. Unfortunately we need the ROS people to sync the new version 1.6.2 and it will probably take few weeks. Until than, you must compile it yourself :(

Just configure `apt` to download pkgs from `ros-shadow-fixed` instead of `ros`. That is the recommended procedure to install staged updates.

See [wiki/ShadowRepository](http://wiki.ros.org/ShadowRepository).

gavanderhoorn, 2018-05-30:
I can confirm that `1.6.2` fixed this.

And it wasn't just an issue with the Kinetic release, Indigo was/is affected as well.

facontidavide, 2018-05-30:
OK

closed 2018-05-30 by facontidavide