PlotJuggler/PlotJuggler #87: PlotJuggler as RVIZ PLUGIN

CesMak, 2018-06-08:
A really nice to have plugin in RVIZ would be PlotJuggler

thus see all the stuff images etc. etc. on one screen! So no one needs to switch between a PlotJuggler and Rviz window :) 

If you would develop such a feature it would be perfect :)

facontidavide, 2018-06-08:
Hi,

First of all thanks for the suggestion, though. It is good to have a conversation about this kind of options and possible improvements.

It was suggested in the past to make PlotJuggler available as [rqt plugin](http://wiki.ros.org/rqt/Tutorials/Create%20your%20new%20rqt%20plugin) 
 but being frank, it is not on my agenda.

I understand the **theoretical** advantages, but I have two reasons for that:

1) PlotJuggler use a lot of real estate on the screen, it is hard to accommodate everything in my opinion. Few of us have a 38 inches screen  ;)

2) PlotJuggler is a desktop applications with it's own plugin. ROS is supported only by plugins and it will never be a hard dependency of the code application.

If someone feels adventurous he/she can try to implement this and share the results with a PR :)

Cheers

Davide

facontidavide, 2018-08-10:
@VictorLamoine even if I am not supposed to help a competitor (just kidding... almost), I would like to suggest taking a look to my other package, **ros_type_introspection**, that allows an application like yours (and mine of course) to subscribe to ANY message and split it in individual time series.

pfankhauser, 2018-08-10:
Sorry to chime in, but if you're interested, our plotting tool rqt_multiplot is available as rqt plugin:

http://wiki.ros.org/rqt_multiplot

-Péter

facontidavide, 2018-08-10:
@pfankhauser 

[image]

VictorLamoine, 2018-08-10:
I have opened an issue about ROS introspection: https://gitlab.com/InstitutMaupertuis/graph_rviz_plugin/issues/11

I don't need it (now) so I won't implement it but if someone creates a merge request I'll be happy to merge it. Thanks for the tip!

facontidavide, 2019-10-09:
This will never happen. Closing the issue

closed 2019-10-09 by facontidavide

weird-bot-ros, 2026-01-03:
How about a rviz panel which can choose whether to load a bag or stream choosing the layout. 

In short the panel can have select mode to either load or stream. Choose a layout from the dropdown in the directory given in the label and click button which launches plotjuggler with these configuration.

I understand that u can always run plotjuggler in a terminal and manually choose layout or stream and select the data to stream. But having rviz panel kind of makes things a bit more convenient that we would generally have different rviz file for different domains like navigation, localization etc and each could save the rviz file accordingly and when time to debug data, open the panel and just click after selecting the configuration.

Is that a weird solution 😅😅 ?