PlotJuggler/PlotJuggler #88: Quicker refresh/restart method for streaming ros topics

alsaibie, 2018-06-20:
Hi, 

When streaming rostopics, would it be possible to refresh the stream update if I restart the nodes, rather than have to stop streaming, restart and reload the topics?

Any pointers to where that can be implemented will be appreciated. 
I have the latest master ver: 1.6.2

Great tool by the way, thanks.

facontidavide, 2018-06-21:
Hi,

I am not sure what you mean by "refresh". 
Do you mean cleared the memory of the buffers (past measurements) ?

alsaibie, 2018-06-21:
Actually, you are right. I'll just keep roscore running in the background to keep the plots alive. Didn't consider that. That solves my concern. I was spawning master off nodes directly.

For a user experience, definitely a reconnect option when roscore went down/restarted would be useful. Or keep have PlotJuggler keep a ros master running? But that may interfere with other ros use scenarios.

facontidavide, 2018-06-22:
I don't think PlotJuggler should keep the roscore running. 
I created a branch in which I started playing around with some code changes https://github.com/facontidavide/PlotJuggler/tree/issue_88

My plan is to detect that the roscore was killed using **ros::master::check()** and ask the user if the plugin should be stopped or else...

I think I will have something usable the next week.

facontidavide, 2018-06-24:
Hi,

ok, I think I have done ;)

Take a look to this video to see how the new feature on the branch https://github.com/facontidavide/PlotJuggler/tree/issue_88 works.

https://vimeo.com/276755814

If it looks okay, let me know, and I will merge it with master

Cheers

alsaibie, 2018-06-24:
Hi, 

Just tried it out. Works like a charm.

Thank you very much!

closed 2018-06-25 by facontidavide

CorentinChauvin, 2018-06-27:
Hello,

I think I found a small bug about it. Here is how I made it crash :

- Start the ROS_Topic_Streamer, but close the dialog without selecting any topic.
- Start the ROS_Topic_Streamer again. At this moment, the dialog doesn't even appear, and PlotJuggler crashes.

Nevertheless, thank you very much for this tool!
Corentin

facontidavide, 2018-06-27:
Thanks for reporting, I will check this.

facontidavide, 2018-06-27:
fixed with last commit. Thanks again