PlotJuggler/PlotJuggler #92: Linker error: undefined reference to `DataStreamer::connectionClosed()'

kartikmohta, 2018-07-13:
Trying to compile the latest from the master branch (c9a5ed9ef458e1b176cd38f2128dd3103eb5dfe2), I get the following error:
```
CMakeFiles/DataStreamROS.dir/DataStreamROS/datastream_ROS.cpp.o: In function `DataStreamROS::timerCallback()':
src/PlotJuggler/plugins/ROS/DataStreamROS/datastream_ROS.cpp:217: undefined reference to `DataStreamer::connectionClosed()'
CMakeFiles/DataStreamROS.dir/DataStreamROS/datastream_ROS.cpp.o: In function `DataStreamROS::~DataStreamROS()':
src/PlotJuggler/./include/PlotJuggler/datastreamer_base.h:27: undefined reference to `vtable for DataStreamer'
CMakeFiles/DataStreamROS.dir/DataStreamROS/datastream_ROS.cpp.o: In function `DataStreamROS::DataStreamROS() [clone .cold.1120]':
src/PlotJuggler/./include/PlotJuggler/datastreamer_base.h:27: undefined reference to `vtable for DataStreamer'
CMakeFiles/DataStreamROS.dir/DataStreamROS_autogen/mocs_compilation.cpp.o: In function `DataStreamROS::qt_metacast(char const*)':
build/plotjuggler/plugins/ROS/DataStreamROS_autogen/L2RW7QHEBF/moc_datastream_ROS.cpp:97: undefined reference to `DataStreamer::qt_metacast(char const*)'
CMakeFiles/DataStreamROS.dir/DataStreamROS_autogen/mocs_compilation.cpp.o: In function `DataStreamROS::qt_metacall(QMetaObject::Call, int, void**)':
build/plotjuggler/plugins/ROS/DataStreamROS_autogen/L2RW7QHEBF/moc_datastream_ROS.cpp:102: undefined reference to `DataStreamer::qt_metacall(QMetaObject::Call, int, void**)'
CMakeFiles/DataStreamROS.dir/DataStreamROS_autogen/mocs_compilation.cpp.o:(.data.rel.ro._ZTI13DataStreamROS[_ZTI13DataStreamROS]+0x10): undefined reference to `typeinfo for DataStreamer'
CMakeFiles/DataStreamROS.dir/DataStreamROS_autogen/mocs_compilation.cpp.o:(.data.rel.ro+0x0): undefined reference to `DataStreamer::staticMetaObject'
collect2: error: ld returned 1 exit status
```
Looks like it's not able to find the `connectionClosed` signal which is defined in `datastreamer_base.h`. I think that the header needs to be added to the source files for the plugin so that cmake generates the files for the signal/slot stuff to work:
```diff
--- a/plugins/ROS/CMakeLists.txt
+++ b/plugins/ROS/CMakeLists.txt
@@ -54,6 +54,7 @@ add_dependencies(DataLoadROS
 SET( DATASTREAM_SRC
     DataStreamROS/datastream_ROS.cpp
     DataStreamROS/datastream_ROS.h
+    ../../include/PlotJuggler/datastreamer_base.h
     )
 
 add_library( DataStreamROS SHARED ${DATASTREAM_SRC} )
```

facontidavide, 2018-07-13:
Hi,

I will apply the change you suggested, it won't hurt.

I just wonder why  Travis compilation seems to work just fine.

kartikmohta, 2018-07-13:
I have `--no-undefined` in my default LDFLAGS which seems to catch such things.

facontidavide, 2018-07-17:
It should be solved. thanks

closed 2018-07-17 by facontidavide