PlotJuggler/PlotJuggler #93: XY Plot fails with data on same topic

ipa-lth, 2018-07-18:
Hi,
I try to track some conjugate complex poles of a system estimation. In this case the real part would be the X-Axis and the imaginary part the Y-Axis.

I publish the numerical data on a own ros msg as seperated values in a float list.

It works ok but suddenly a warning pops up "The creation of the XY plot failed because at least two timeseries don't share the same time axis". This window pops up several times, until it continues for some time.

Since both values come in the same message I don't really understand what is going wrong.
If for some reasons some data points cannot be displayed, I would be fine with it. The modal püopup however prevents me from actively using XY-plot.

I am using Tag 1.6.5 (newer version did not compile but that is another problem)

I attached a bag file with some of my data, an xml layout and a screencast video 
[XY_issue.zip](https://github.com/facontidavide/PlotJuggler/files/2205052/XY_issue.zip)

Thanks!

facontidavide, 2018-07-18:
Thanks for reporting. I will look at it as soon as possible.

facontidavide, 2018-07-19:
Ok, I think I fixed it. 

It was an important thread safety bug that affected only the XYPlot case.

Thanks for reporting and for sharing XY_issue.zip. it made my degubbing work much easier.

Davide

closed 2018-07-19 by facontidavide