PlotJuggler/PlotJuggler #94: Autoscroll

TobyUllrich, 2018-07-20:
How do i turn on the Autoscroll during streaming. Changing the buffer Time in the bottom right corner does nothing.

And I get this message when trying to stream a message with 6 3x1 lists:
"terminate called after throwing an instance of 'std::runtime_error'
  what():  Buffer overrun in RosIntrospection::ReadFromBuffer
Aborted (core dumped)"

facontidavide, 2018-07-20:
Please share the ROS bag to reproduce the issue.

TobyUllrich, 2018-07-20:
Here is the bag
[2018-07-20-19-27-03.bag.zip](https://github.com/facontidavide/PlotJuggler/files/2214844/2018-07-20-19-27-03.bag.zip)

facontidavide, 2018-07-20:
This is really weird... I am trying to decode the message... is this the message definition?

```
int32[] position
int32[] speed
int32[] torque
int16[] drivertemperature
int16[] motortemperature
int8[]  error
```

TobyUllrich, 2018-07-20:
Yes this is correct each list has a length of 3 for the 3 motors I use

facontidavide, 2018-07-20:
if you display bag you sent me with with __rostopic echo__ or __rqt_plot__ does it work?

facontidavide, 2018-07-20:
Ok, unfortunately I am about to give up.
I added a unit test for your particular Message type and it works just fine.

In your particular case, a message should have size 75 bytes, when it contains 3 elements per field.
Even if a message is COMPLETELY empty, the size of the message should be 18 bytes.

What is weird is that your rosbag contains 28 messages. The first 21 messages have size 75 es expected, whilst the following on have size

0
3
0
0
768
0

This doesn't make any sense from my point of view :(
I am afraid that I am not able to help you, the message seems to be "corrupted"

facontidavide, 2018-07-22:
I really would like to help you, but I have no clue :(

As I saif, if you check the rosbag with this file, we don't get what we expect.

```c++
#include <ros/ros.h>
#include <rosbag/bag.h>
#include <rosbag/view.h>

// usage: pass the name of the file as command line argument
int main(int argc, char** argv)
{
    if( argc != 2 ){
        printf("missing filename\n");
        return 1;
    }
    rosbag::Bag bag;

    try{
        bag.open( argv[1] );
    }
    catch( rosbag::BagException&  ex)
    {
        printf("rosbag::open thrown an exception: %s\n", ex.what());
        return 1;
    }

    rosbag::View bag_view ( bag );

    int count = 0;
    for(rosbag::MessageInstance msg_instance: bag_view)
    {
        const std::string& topic_name  = msg_instance.getTopic();
        const size_t msg_size  = msg_instance.size();
        printf("%s [%d]: message size %d\n", topic_name.c_str(), count++, msg_size );
    }
    return 0;
}
```

As I said, the first 22 messages are fine, but the last 6 are not.

I am afraid that I don't know how to help you. I added a unit test [here](https://github.com/facontidavide/type_introspection_tests/blob/master/tests/deserializer_test.cpp#L458), but it seems to work...

Sorry.

TobyUllrich, 2018-07-22:
Ive solved it. It was indeed a corrupted message. It works fine now

closed 2018-07-22 by TobyUllrich

facontidavide, 2018-07-22:
I am happy to hear that it is solved :)