PlotJuggler/PlotJuggler #100: Floating point truncated exception

twostripe, 2018-08-10:
version: **1.7.1**
os: **ubuntu 16.04**

When I tried to load any large rosbag files through PlotJuggler, and select all the topics, I get

> The plugin [DataLoad ROS bags] thrown the following exception: 
> Floating point truncated

[image]

Also getting the following error from the console at the same time:

> Not monotonic time in [ /rosout/header/seq ]:  1.53273e+09  /  1.53273e+09

Maybe they're related.

twostripe, 2018-08-10:
I've isolated this error to only when pulling from a particular message type:

```Header header
uint64 trace_id_upper
uint64 trace_id_lower
string node_name
TracerSpan[] spans
```

which contains

```## Tracer Span ##
uint64 span_id
uint64 span_parent_id
TracerSpanRef[] span_refs
time start_time
time end_time
string operation_name
bool is_root
uint8 MESSAGE_SEND=1
uint8 MESSAGE_RECV=2
uint8 LOCAL=3
uint8 span_type
KeyValue[] tags
```

which in turn contains

```##TracerSpanRef##
uint64 span_id
uint8 CHILD_OF=1
uint8 FOLLOWER_OF=2
uint8 ref_type
```

facontidavide, 2018-08-12:
Please share a rosbag that I can use to reproduce the error and I will fix it.

facontidavide, 2018-08-15:
Thanks for sharing the file. The mystery is "solved" but I am not sure what to do about it.

Timeseries in PlotJuggler are represented using timepoints which are double/double pairs (time/value).

In your rosbag there __/commander/trace_msgs/spans.0/span_id__ has the value 9943996159296244680

When converted to double, this number become 9943996159296244000 because of numerical cancellation.
Since, IMHO, no data is better than wrong data, I check this kind of problem and launch an exception.
I may have a more informative exception, though. 

I am not even sure if it is really possible to fix this, since the chart library that I am using, Qwt, seems to convert everything to double itself, sooner or later. This means that, even if I fix numerical cancellation on my size, it will happen on the Qwt side, silently, without any warnings.

May be I am wrong and there is a solution modifying something inside Qwt itself.

But of course, that is a LOT of work and I am not sure that I should focus on this issue, unless nore people report this and I find out that this is a frequent problem.

This doen't mean that I won't _try_ fixing it, it just mean that I am prioritizing other issues.

twostripe, 2018-08-15:
Those messages aren't particularly useful for me. My ideal simple case would be to have the program throw an error (ideally name the errant topic), but still be able to load the other data.

facontidavide, 2018-08-15:
Good point, I will have a different kind of behavior when this particular case is detected

facontidavide, 2018-08-15:
I think that the commit eb474776b2798783476413dd2bc4037a4f972ea8 is sufficient for the time being.

I am not 100% proud of it but at least it is far better than before.

facontidavide, 2018-08-16:
I will reopen this and mark it as enhancement because I want to keep this issue in the back of my mind...

twostripe, 2018-08-17:
Now getting this new error modal:

[image]

Which is more helpful, but ultimately unclear how big of a deal it is. At minimum, it would be nice to know what topic(s) was to blame for this.

But ultimately much better.

closed 2020-08-07 by facontidavide