PlotJuggler/PlotJuggler #103: Window does not respond

twostripe, 2018-08-10:
version: **1.7.1**
os: **ubuntu 16.04**

Occasionally PlotJuggler becomes unresponsive and I have to force quit. Often this happens when I've loaded large files or are otherwise doing "two things at once".

No error message is visible. I'll wait as much as 5 min then force quit.

[image]

twostripe, 2018-08-10:
"Do you want to remove previous data" tends to do this, but it recovers. Perhaps a loading bar for this action would be useful.

facontidavide, 2018-08-12:
This sounds more like a deadlock, but I might be wrong, of course. 
There should already be a loading bar, see here

 https://github.com/facontidavide/PlotJuggler/blob/master/plugins/ROS/DataLoadROS/dataload_ros.cpp#L134

How large is the rosbag? May be we are running out of physical memory (RAM) and using the swap... can you check how much CPU and RAM Plotjuggle is using in this case, using the command __top__ or __htop__ ?

If it is not a memory problem, I am afraid that I will need to ask you to share the rosbag with me to see if I can reproduce the error.

twostripe, 2018-08-14:
**During Loading**
When loading _all the data_ from very large files (~20GB), PlotJuggler definitely takes up ~100% CPU, though only ~10-25% MEM.

I see the loading bar while loading. Everything is fine, this grey-out doesn't occur.

**After Loading**
After the loading bag gets to 100 and after going through all the dialogues the grey-screen happens. Using 50% RAM (8GB) and 12% CPU.

Stayed this way for ~15 min, then i could interact with it. When I search for things (or clear the searchbar) it greys-out for another minute or three. Now that I know that it's still working I can still use it, but perhaps the UI can be changed (minimal loading bar in the main window itself?) so it doesn't lock up like this.

facontidavide, 2018-08-15:
I guess that I have never detected this because the biggest rosbag I analyzed was 6-10 Gb ;)

Please try the very latest version of PlotJuggler on __master__ since I pulled yesterday a couple of optimizations... I might accidentally fixed your issue.

I will try to create a huge rosbag to see if I can reproduce the issue and see if there is any opportunity for optimization. But that would take me few days.

If you are eager to help, I can tell you how to find the issue yourself using the tools that I would use myself ;)

Ladies and gentleman, please welcome Hotspot (https://github.com/KDAB/hotspot/releases/tag/v1.1.0).

This is the application I use to profile performance issues.

What you can do with it is to attach to a running PlotJuggler process (Record option) and profile it during the freeze. In this way you get a complete report of where the time is spent (flamegraph)

Cheers

facontidavide, 2018-08-15:
The more I think about it, the more I believe that something is happening under the hood that is NOT related to the life parsing itself. Actually I am pretty sure that is related to the class FilterableListWidget...

I know it is a pain in the ass, but is there any way to share this big rosbag?

facontidavide, 2018-08-15:
Forget about sharing the rosbag or using yourself hotspot, I found the issue.

[image]

Commit coming soon... I can tell you in advance what the problem is: FilterableListWidget::addToCompletionTree is too expensive.

This is needed only when the following filtering mode is active 
[image]

twostripe, 2018-08-17:
No longer seeing grey-screen. Things seem to load faster and be more responsive as well. Not sure if that is placebo or not.

twostripe, 2018-08-17:
Spoke too soon. I see the grey screen again.

[image]

**Steps to Reproduce**
1. Load a file with lots of data ~20GB, load all topics
1. Load another file with lots of data ~20GB, load all topics
1. "Do you want to remove previously loaded data?" `Yes`

I left it running for ~10 min and no change. No significant spike in memory or CPU.

twostripe, 2018-08-17:
> Mark,
>
> I am sorry to hear that you still have problem and I really want to help,
> but I am afraid that I don´t know how.
> Not all the rosbag are created equal. From the point of view of
> PlotJuggler, the cost of loading can be dramatically different according to
> the type of message.
>
> My related commits removed a big bottleneck in the FilterableListWidget
> class, but right now, based on your description, it is almost impossible
> for me to debug your issue.
>
>    - it may be that you simply run out of memory (not surprisingly, since
>    you are loaginf 40 Gm of stuff in memory, with quite a lot of memory
>    overhead). On the other hand you mentioned that you checked memory and that
>    is OK.
>    - or there is still some inefficient code *somewhere*.
>
> *I am really eager to help, but I can hardly do it* without either:
>
>    - your rosbag,
>    - the results from Hotspot / perf
>
> Maybe I will have some "eureka" moment and find the error, but I can't
> promise that without further information.

That's alright. It's not good practice for me to load the *entire* rosbag
with all of its topics. If this becomes an actual blocker for me I'll
construct some sample code and send it your way.

Thank you.

facontidavide, 2018-08-18:
I have been thinking and I have an hypothesis...

What if:

1) You load the first file
2) You remove everything calling File -> Delete All Data
3) You load the second file.

The final result should be the same, but __maybe__ you want experience the freeze. Let me know

facontidavide, 2018-08-18:
I am proud to say that it is fixed on branch 1.8.x
The problem could be reproduced with a very high number of timeseries, it was NOt related to the size of the data itself or to the amount of RAM.

It was again a problem with class FilterableListWidget...

Happy weekend

closed 2018-08-18 by facontidavide

twostripe, 2018-08-18:
!! Congrats