PlotJuggler/PlotJuggler #105: Loading Data Requests

twostripe, 2018-08-10:
version: **1.7.1**
os: **ubuntu 16.04**

The following are some UI changes and/or bugs for the Load Data process:

_These are mostly encapsulating the Dialog (called Dialog) that pops up while loading data_

**Bugs**
1. closing the window with the [x] or [cancel] buttons seems to load the data anyway
1. as the data is loading, if you hit [cancel] the warning appears: "Do you want to remove the previously loaded data". It's unclear why.

**Enhancements**
1. it it would be nice to have a button to unselect all the topics
this could be even a command like [Ctrl]+[Z], or even double-clicking just selecting a single topic. perhaps two buttons, one to select and one to deselect is the answer here.
1. it would be nice to search for topics
1. it would be nice if the `Maximum size of arrays` could be larger. I'm reaching this limit with bags larger than a few minutes with certain topics. It's also unclear what happens when this limit is reached, is that data unusable?
1. it would be nice if the `Maximum size of arrays` could be set globally, or at least persist each time the dialog opens. Or perhaps an option for "no maximum size"

facontidavide, 2018-08-12:
Just pushed a fix to BUG 1.

I will investigate BUG 2 in the next days.

About the enhancements:

1) Ok

2) Sure

3) Not exactly. Let suppose that you load topic_A first and then topic_B from the same file later on. Just answer "NO" to the question "Do you want to remove the previously loaded data", that in your case would be the already loaded topic_A. Considering the way in which the information is stored in a rosbag, I can hardly think how to make this process faster.

4) I agree with you that it is unclear and some documentation is needed. The big problem is what is: what is an array? If it is something relatively small like joint_state, no problem, but what is the array is a huge map, pointcloud, image or the data from a velodyne (tens of thousands elements in the array). Any single pixel (element of the array) would have its own timeseries. this makes no sense and would also use a crazy amount of memory. I want to prevent the user from loading this kind of huge arrays, because most of the time, they don't really know what they are doing.
Many people will just "load everything".

5) As I said in point 4), I will never remove this value, but I agree with you that it can be remembered. I will do that.

facontidavide, 2018-08-12:
About enhancement 4

I am realizing that you might erroneously think that the size of the array is related to time.
It is __not__. A single time series might have a million points.
That size is related to the number of elements in a __single__ message type.

facontidavide, 2018-08-13:
I think that I more or less fixed all the bugs/enhancements you included in this issue.

Let me know what you think of these changes.

Cheers

closed 2018-08-13 by facontidavide

twostripe, 2018-08-17:
**Bugs**

> 1 closing the window with the [x] or [cancel] buttons seems to load the data anyway

Fixed. Except the terminal still displays the message:

`The loading operation took 2161 milliseconds`

which seems misleading, if not incorrect.

> 2 as the data is loading, if you hit [cancel] the warning appears: "Do you want to remove the previously loaded data". It's unclear why.

Fixed.

**Enhancements**

> 1. it it would be nice to have a button to unselect all the topics
> this could be even a command like [Ctrl]+[Z], or even double-clicking just selecting a single topic. perhaps two buttons, one to select and one to deselect is the answer here.

Fixed, I like it.

> 2. it would be nice to search for topics

I don't see this. It's not a crucial enhancement

> 3. It would be really nice to be able to load individual topics from a file I already loaded. Say I load a bag and select a topic, then realize I also need another topic. I have to load the whole bag again.

> _Not exactly. Let suppose that you load topic_A first and then topic_B from the same file later on. Just answer "NO" to the question "Do you want to remove the previously loaded data", that in your case would be the already loaded topic_A._

Noted.

> 4. it would be nice if the Maximum size of arrays could be larger. I'm reaching this limit with bags larger than a few minutes with certain topics. It's also unclear what happens when this limit is reached, is that data unusable?

Noted. Thank you for adding the tooltip.

> 5. it would be nice if the Maximum size of arrays could be set globally, or at least persist each time the dialog opens. Or perhaps an option for "no maximum size"

I see it remembering.

Thank you