PlotJuggler/PlotJuggler #99: Loading data doesn't refresh plots (curves)

twostripe, 2018-08-10:
version: **1.7.1**
os: **ubuntu 16.04**

Loading data new data _sometimes_ doesn't refresh all the curves. I noticed it most often when there are more plots or large data, but that could be a placebo.

Anything that re renders the curve fixes this (changing tabs, sliding the slider, altering the curve).

The axis seem to always change, but not the curve.

facontidavide, 2018-08-12:
I believe that I fixed this in commit afc570926659f8087d6574a4d19b9297dcd36d88
Therefore 1.7.2 should be ok. Please let me know if the problem is still there or not

facontidavide, 2018-08-12:
I will temporary close this, since I am positive that it is fixed. Feel free to reopen it if you detect that the bug is still there.

closed 2018-08-12 by facontidavide