PlotJuggler/PlotJuggler #106: Improvement: Font size adjustment

michael1874888, 2018-08-20:
Hi @facontidavide ,
I think it would be great to be able to change the font size of the legend. In some cases (e.g. without any subplot),  the font size of the legend looks small.

facontidavide, 2018-08-20:
Sure, I will add this feature in the future, but probably not in PlotJuggler 1.X

Thanks for the suggestion

facontidavide, 2019-01-24:
You can now resize the legend using **CTRL + Mouse Wheel.**

[image]

closed 2019-01-24 by facontidavide

michael1874888, 2019-01-25:
It looks great!

idref, 2021-11-17:
Hi @facontidavide,
first of all, thx for this great tool !

I think it would be nice to be able to change the font size of the annotation of the axes and maybe the thickness of the lines as well. With many subplots the fonts are very small.

Is there a current way to do this?
Thank you !