PlotJuggler/PlotJuggler #109: Segmentation fault

jihoonl, 2018-08-29:
It crashed when I attempted to remove >10 topics from left panel including active curves.

```
jihoonl@whoola:~$ rosrun plotjuggler PlotJuggler 
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
Stack trace (most recent call last):
#22   Object "", at 0xffffffffffffffff, in 
#21   Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x473508, in _start
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f0d6511f82f, in __libc_start_main
      Source "/build/glibc-Cl5G7W/glibc-2.23/csu/../csu/libc-start.c", line 291, in __libc_start_main [0x7f0d6511f82f]
#19   Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x472dd7, in main
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f0d65ff5beb, in QCoreApplication::exec()
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f0d65fedb49, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f0d660467ce, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#15   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f0d647ae49b, in g_main_context_iteration
#14   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f0d647ae3ef, in g_main_context_dispatch
#13   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f0d647ae196, in g_main_context_dispatch
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f0d674231ff, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f0d66317f07, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f0d66334184, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f0d6632f15d, in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f0d65ff038a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0d668e8515, in QApplication::notify(QObject*, QEvent*)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0d668e305b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0d66942b62, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f0d65ff038a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
#3    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0d668ea09a, in QApplication::notify(QObject*, QEvent*)
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0d668e305b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#1    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0d669262f6, in QWidget::event(QEvent*)
#0    Object "/opt/ros/kinetic/lib/plotjuggler/PlotJuggler", at 0x4aa467, in FilterableListWidget::removeSelectedCurves()
Segmentation fault (Address not mapped to object [(nil)])
Segmentation fault
jihoonl@whoola:~$ 
```

facontidavide, 2018-08-29:
Sorry about that and thanks for reporting. I will look at it once I am back from my holiday ;)

facontidavide, 2018-09-08:
This is similar to a bug I already fixed. 

1) which version of PlotJuggler are you using?
2) Is it possible to reproduce the bug using the dummy data? (Launch the application with argument "-t").

facontidavide, 2018-09-13:
Ok, I suppose this was fixed then... otherwise, feel free to open it again

gvallicrosa, 2018-09-14:
Segmentation Fault in 1.8.3 when trying to change a plot to XY with the right-click menu.
Ubuntu 18.04, ROS Melodic.
Can be reproduced with -t or even without data.

    Stack trace (most recent call last):
    #31   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f065106c68b, in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*)
    #30   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55666989f67f, in PlotWidget::eventFilter(QObject*, QEvent*)
    #29   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55666989ead0, in PlotWidget::canvasContextMenuTriggered(QPoint const&)
    #28   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651f682af, in QMenu::exec(QPoint const&, QAction*)
    #27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f065106a9e9, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
    #26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f06510c58ee, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
    #25   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f064e7e254b, in g_main_context_iteration
    #24   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f064e7e24bf, in g_main_context_dispatch
    #23   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f064e7e2286, in g_main_context_dispatch
    #22   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f0649abe1bf, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
    #21   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f06516072ea, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
    #20   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f065162fff4, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
    #19   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f065162e522, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
    #18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f065106c9a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    #17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651df20f3, in QApplication::notify(QObject*, QEvent*)
    #16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651dea82b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    #15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651e467b9, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
    #14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651e43e84, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
    #13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651df1621, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
    #12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f065106c9a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    #11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651df264e, in QApplication::notify(QObject*, QEvent*)
    #10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651dea82b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    #9    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651f6d6ba, in QMenu::event(QEvent*)
    #8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651e29037, in QWidget::event(QEvent*)
    #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651f6b392, in QMenu::mouseReleaseEvent(QMouseEvent*)
    #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651f6a56a, in QMenu::leaveEvent(QEvent*)
    #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651f630bb, in QMenu::setIcon(QIcon const&)
    #4    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651de680b, in QAction::activate(QAction::ActionEvent)
    #3    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f0651de4121, in QAction::triggered(bool)
    #2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f065109b6ce, in QMetaObject::activate(QObject*, int, int, void**)
    #1    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55666989dc64, in PlotWidget::on_convertToXY_triggered(bool)
    #0    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x5566698aaace, in TimeseriesQwt::updateData()
    Segmentation fault (Signal sent by the kernel [(nil)])
    Segmentation fault (core dumped)

facontidavide, 2018-09-14:
This is a completely different kind of problem. I will try to fix it ASAP

facontidavide, 2018-09-14:
just pushed the fix into the branch master! Are you able to try it?

gvallicrosa, 2018-09-17:
Done. You can check it here:
https://drive.google.com/file/d/1ARUhgvTwyzxrs6WIx2_wvrIVJADagIJq/view?usp=sharing

facontidavide, 2018-09-17:
First: thank you so much for the time you spent to help me solve this issue.
Second: that was an embarassing error that happens when you compile in Release but not in Debug. A variable was not initialized.

It should be ok now.

closed 2018-09-17 by facontidavide

gvallicrosa, 2018-09-17:
Confirmed that it works if data is dragged to the x axis with right click and then other data is added as usual. If right-clicking without data, XY plot is greyed out. Missing the nice popup explaining how to make the XY plots that we had in 1.7.1.

facontidavide, 2018-09-17:
That is intentional. I will probably substitute the PopUp message with a ToolTip message.

gvallicrosa, 2018-09-18:
Thanks for the fast reply and issue solving :smiley: