PlotJuggler/PlotJuggler #110: Can I set min/max on X axis for XY streaming plots?

atgidev, 2018-08-29:
I'm new to PlotJuggler, so perhaps I missed this but is there a way I set min/max limits on the  X-axis for XY streaming plots?

I'm using the app in streaming mode to plot data from a ROS topic message containing X-Y coordinates from a multi-node ROS system that calculates the position of a device by triangulating range data from multiple ultra-wideband precision location sensors.

It seems I have everything working except setting limits on the X-axis. If I could understand how to do this, I think the app would work great for our needs.

Btw - kudos on the work, the problems you solved here took a non-trivial amount of effort.

facontidavide, 2018-08-29:
Unfortunately this option is not currently available. Implementing it will take me some time tough. I will implement it once other issues are closed

JimmyDaSilva, 2021-01-20:
I would need this as well.

facontidavide, 2026-10-01:
Closing this PJ3 report about fixed X-axis limits for streaming XY plots during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide