PlotJuggler/PlotJuggler #112: Log scale plotting?

ahundt, 2018-08-31:
Is it possible to plot data on a logarithmic scale? I've got a time series that decays exponentially, so the Y values go down from 1 to 10^-3 to 10^-6 as time proceeds linearly. Non-log plots just look like the value drops to zero and stay there. :-)

ahundt, 2018-08-31:
ok cool thanks, enjoy your holiday!

facontidavide, 2019-10-09:
I think this will serve only a tiny fraction of my users and I will probably not implement it. I am temporarily closing this issue.

closed 2019-10-09 by facontidavide

gitterspec, 2022-01-30:
Also very interested in this feature. Wondering if it might end up back on the roadmap?

edward9503, 2022-08-07:
Hi @facontidavide ,

I think there is a growing tendency for using the logarithm scale in streaming. It would be great if this feature can be added. Many thanks.

sjs329, 2022-09-02:
+1 for log scale plotting!

sneuro, 2022-12-17:
+1
Required function!

naokiuchida, 2023-01-31:
+1 necessary feature!

andrei-markov-emlid, 2023-08-15:
+1 Very useful feature

facontidavide, 2023-08-15:
PR welcome