PlotJuggler/PlotJuggler #115: Feature request: angle gauge view

llvs, 2018-10-24:
Dear Davide,
First: PlotJuggler is great and I like it a lot.
It would be great if PlotJuggler had an angle gauge view, like a horizon line and a compass.
Something like [this](https://www.hackster.io/Aritro/diy-flight-instruments-for-horizon-and-compass-43e2f6) even with a lot less eye candy would be sufficient.
It would of course be a new type of plot.
Thanks for reading this!

facontidavide, 2018-10-24:
Thanks, I am happy that you found PlotJuggler useful.

Good news:

If you don't mind having the gauge view in a separate window, it is very easy, just use the the StatePublisherPlugin. 
https://github.com/facontidavide/PlotJuggler/blob/master/include/PlotJuggler/statepublisher_base.h
You can display the value corresponding to the time of the vertical tracker.

Bad news ...

this is one of those "nice to have" features that in my opinion, very few users need, therefore it falls in the category of "not a priority, but I can do it if sponsored".
Don't take me wrong, I always try to implement features that people want, but I prioritize those that cover more use cases or the sponsored one.

I will keep this issue open, though. Maybe at some point, I will find the time to implement it :)

If you want to do it yourself,

llvs, 2018-10-24:
I fully understand that you are prioritizing, it is voluntary work on open source software after all.
I will look into the StatePublisherPlugin to see whether that also helps me.
Thank you for that quick answer!

facontidavide, 2019-01-18:
I have found this that may help... still I have no bandwidth to work on this feature

https://github.com/Berrima/Qt-custom-gauge-widget

llvs, 2019-03-22:
Just as an update, because you said in another thread that you use the Qwt lib for charting.
Qwt seems to already bring a compass and a horizon gauge with it:
https://qwt.sourceforge.io/controlscreenshots.html
Maybe, it is less work than you thought.

facontidavide, 2019-03-22:
It is probably not much work. I am just too busy with a ton of other features to work on this right now.

Cheers

llvs, 2019-03-22:
That's perfectly fine, I just wanted to make sure that the wheel is not invented twice and that you know that the building blocks might already be available in Qwt.
Take your time to implement whatever you deem most important! Thank you for all the work that goes into PlotJuggler!

Cheers

facontidavide, 2020-04-29:
not going to happen

closed 2020-04-29 by facontidavide