PlotJuggler/PlotJuggler #116: Feature request: Multiply/Offset axis ticks

llvs, 2018-10-25:
Dear Davide,
I have again a feature request: It would be very handy to be able to apply some kind of multiplication / offset removal to the values displayed on a per-plot basis.

Example 1: An angle value is given in radians and I want to have it displayed in degrees, i.e. I want to apply the transform "*180.0/pi" to the values.

Example 2: A value starts at large values and changes not so much from then on and I am only interested in the relative changes, not so much in the absolute value. Therefore, I would like to be able to remove the offset from the first value of the plot so that the plots start at zero.

Best regards

facontidavide, 2018-10-25:
So you mean a offset in the Y axis (value) not the X Axis (time).

That will be possible once I merge the new branch https://github.com/facontidavide/PlotJuggler/tree/custom_math

You will be able to apply simple formulas that modify the value, being an offset the simplest one. Let me show you.

[image]
[image]

llvs, 2018-10-26:
Thank you! I will try that code branch.

facontidavide, 2019-01-15:
Hi,

with the nex "custom plot" functionality, it is now possible to do what you mentioned.

Cheers

closed 2019-01-15 by facontidavide

llvs, 2019-01-15:
Thanks a lot!