PlotJuggler/PlotJuggler #118: Feature request: equal axis scaling

llvs, 2018-10-26:
Dear Davide,

Here comes another feature request:
When using a custom x-axis (dragged with the right mouse button), I can create an x,y position plot which is very nice. However, it would be very useful to have an option for a locked axis ratio such that, e.g. right angles, appear as such in the plot. Like with the Matlab command "axis equal".

Best regards

facontidavide, 2018-10-26:
It would be a nice feature to have. I will probably implement it at some point

facontidavide, 2019-02-06:
From a UI point of view, I am not sure how and where this feature should be toggled... Suggestions?

koenlek, 2019-03-01:
@llvs great suggestion! We would really like this feature too. 

@facontidavide: How about in the right-click menu on the plot? 

How about adding it to the zoom related section, the new section would become:
- Edit X-axis limits (gray out or hide in case not in x/y plot mode)
- Edit Y-axis limits
- Set X/Y scale equal (gray out or hide in case not in x/y plot mode)
- Zoom out
- Zoom out horizontally
- Zoom out vertically 

I checked and both matplotlib and matlab use 'equal' as the terminology for such axis scaling. 'square' instead forces the plot to be square in size (the canvas of the plot, so to say). 

By the way: In the suggestion above I sneaked an extra feature request in: allow setting the X axis limits for X/Y plots. I see that this was already requested in #110. Setting both X/Y axis limits manually actually can break an equal scale. In equal scale mode, I'd just find the square that fits both the chosen x limits and y limits. I.e. take `axis_padding = max(ymax-ymin,xmax-xmin)/2` and if y interval is larger, do `xmin_new = xmin - axis_padding` and `xmax_new = xmax + axis_padding`, if x interval is larger, same idea but sway x/y varnames.

skohlbr, 2019-03-12:
Just tried the XY plot functionality, and this was the first thing that I immediately looked for in the UI after seeing the nice (but off axis ratio) plot. So another vote for this. For most XY data this would be the preferred default methinks.

facontidavide, 2019-03-23:
started working on this feature... It is more hard that I initially estimated :(

facontidavide, 2019-03-24:
I can proudly announce that the "equal axis scaling" as been merged.

There is a fairly high possibility that I broke something in the process, so: "if you see something, say something".

Cheers

llvs, 2019-03-25:
Thanks a lot! Will try it immediately.

aeudes, 2019-03-25:
Cool, it's still just amazing the pace at which this tool evolved !
Very quick testing show strange behaviour with zoom when equal axis is activated, it could be replicated in a graph with angular_velocity.x/y from imu in tilt_laser.bag:
- when using window zoom, sometimes, making successive zoom produce a zoom out
- using mouse wheel zoom makes zoom out unpredictably.
- sometimes, I loose the ability to make a zoom window (click + move did not draw the rectangle, but I see the icon changed)
- i don't known what should be the behaviour of wheel on axis with this mode activated.

facontidavide, 2019-03-25:
Can you share a video (screencast) of the behavior you consider "weird"?

- About "mouse wheel on axis", that will break the __equal ratio__, so, I need to disable it.

- "sometimes, I loose the ability to make a zoom window" this happened to me as well. I am not able to reproduce it unfortunately, I need to fix this!

aeudes, 2019-03-25:
1/ continuous track-pad zoom
2/ 2 successive window zoom

[image]

facontidavide, 2019-03-25:
can you share the rosbag please?

facontidavide, 2019-03-25:
I just pushed a commit that might help.

aeudes, 2019-03-25:
I have used: the sample data with angular_velocity.x/y from imu topic  tilt_laser.bag.

facontidavide, 2019-03-25:
Check the last version

aeudes, 2019-03-25:
With last version both the zoom error seems to be fixed !

I still trigger the "loose the ability to make a zoom window" :
- if you zoom windows to have one point or less.
- zoom out with mouse wheel to maximum
-> loose the zoom window
- zoom in with the wheel, at some point recover the zoom window

facontidavide, 2019-03-25:
fixed... hopefully :)

aeudes, 2019-03-25:
Look like fixed here also !

facontidavide, 2019-03-25:
I refactored this feature to remove some complexity. It should still work fine

closed 2019-03-25 by facontidavide

skohlbr, 2019-03-25:
XY plots have correct aspect ratio per default now, which is awesome. :)
Unfortunately, the separate axis scaling from #135 (X or Y key + mouse wheel) doesn't work anymore (in any plot, not just XY) after updating to latest master.

facontidavide, 2019-03-25:
Check the cheatsheet. There is a new way to do single axis scaling